meshtastic/firmware #1055: Update BLE Pin Pairing code to use standard Nimble BLE methods

garthvh, 2022-01-03:
It the custom pin pairing code we are currently doing to connect to BLE devices is causing some crashing / inability to pair BLE 4.0 ESP32 devices on iOS and recent android versions (10+)

**To Reproduce**
 
Forget the device and attempt to pair a tbeam with the iOS app.

It seems like this update may be a good time to set up just works and pin based pairing for all devices. And settle on a default (I would vote for no pin as default.)

mc-hamster, 2022-01-04:
The cause of this may be similar to the Android pairing problems we've been seeing.

mc-hamster, 2022-01-04:
I had a device that couldn't be discovered by iOS or android. After erasing the flash and reinstalling firmware, the device worked with ble again.

This suggests the problem may not only be with pin pairing.

garthvh, 2022-01-04:
Had it worked previously or was it a brand new device?  In my experience only devices with PINs do that strange disappearing devices that use just works don't really seem to have any details saved in iOS at least.

mc-hamster, 2022-01-04:
I don’t use clients enough to be able to answer that question.

mc-hamster, 2022-01-07:
Disabling MITM made pairing reliable but had other problems.

thebentern, 2022-05-14:
@garthvh is this solved on 1.3 with the new bluetooth platform? If so, I'd like to close it.

garthvh, 2022-05-14:
Seems to be solved in 1.3

thebentern, 2022-05-14:
Closing. We can reopen a new issue if anything crops up on 1.3.

closed 2022-05-14 by thebentern