meshtastic/firmware #1057: Request reset location data when disabling fixed-position

jaykayenn, 2022-01-03:
Docs say to set fixed_position=false to unset, but this doesn't actually clear the coordinates on device, only allows the location to be updated going forward. However, for devices like TLora32 without GPS, the location never gets updated and the last fixed_position coordinates get stuck forever. 

- setlat/lon/alt on device. fixed_position is automatically set to true
- set fixed_position false
- previous location remains in nodeinfo

Expected: clear nodeinfo position when fixed_position=false

Firmware 1.2.50
Device: TLora32 v2.1-1.6

garthvh, 2022-01-24:
I just tested this on TLora32 v2.1-1.6 devices with firmware 1.2.52, it works as expected.  Once fixed_position is set to false an empty position with battery level is sent as the position from the mesh.

Setting fixed_position does not automatically unset the previously set lat/long values it just stops broadcasting the fixed position values.

garthvh, 2022-01-24:
This may be a display issue in the python CLI, the device level data is correct.

closed 2022-01-24 by garthvh