meshtastic/firmware #1071: Add configuration settings to turn off Tbeam BLE "Device is connected but sleeping" functionality by default.

garthvh, 2022-01-07:
Currently the default behavior for t-beams is to turn off the BLE radio for a period of time to "sleep" and save some power. 

This functionality forces users to push a button on the device before being able to send messages in android or connect to the device on iOS.

BLE is low energy, and not not have a "sleeping" state in the spec.  The way the android app is scanning constantly and re-connecting to the device is considered a privacy issue by iOS.

I propose we disable this functionality by default and allow people to enable it if they find value in the marginal gains in battery life valuable.

ianmagnone, 2022-01-07:
I would agree. My use case is mainly camping/hiking where the TBeam will be clipped to a backpack so out of reach. Needing to press a button to enable it to talk to an iPhone to send/receive messages makes it a lot less useful.

thebentern, 2022-01-07:
I concur.
My experiences:
1. The battery life gains to be had on a reasonably periodically sleeping T-Beam are far from impressive vs running with is_always_powered. 
2. They also don't always seem to reliably deliver on the process of queued messages.

tropho23, 2022-01-07:
I concur, and appreciate the detective work to determine this is the source of the problem.  My experience mirrors the others described here, and the extra battery life would indeed be preferable.  I also plan to place my T-beam units in a hard-but-not-impossible-to-reach area on my backpack, adjacent to a solar panel so having to press a switch periodically just to receive or send messages would make it not worth using without enabling _is_always_powered_.

mc-hamster, 2022-01-10:
Any ideas of what to call this setting?

The default value must be "false".

is_power_saving
is_saving_power
is_high_power

We need to pick something that would make sense and be different from is_always_powered and is_low_power .

mc-hamster, 2022-01-10:
No toggle yet, but here's a build with light sleep disabled:

http://www.casler.org/temp/firmware_1.2.50_f302166_no_lightsleep.zip

ianmagnone, 2022-01-10:
I think is_power_saving makes the most sense.

mc-hamster, 2022-01-10:
This is done

closed 2022-01-10 by mc-hamster

mc-hamster, 2022-01-11:
Device under test: tbeam 1.1 w/ ne06m
Voltage: 3.60, powered into battery terminals
Environment: 7 nodes, forwarding mesh packets for each other

New default config:
3.75wh
1.040ah

Power saving:
2.978wh
0.828ah

ianmagnone, 2022-01-11:
I think that's acceptable for having the device be more reliable for the user.