meshtastic/firmware #1079: Freeze after a certain amount of time or traffic

kingoffoxes, 2022-01-11:
Device freezes after a certain amount of time or traffic on Shortfast. (suspect traffic count as i think this happened before but took a lot longer on LongSlow)

3 T-Beams in mesh all using same firmware version set to short fast and all on the same channel.  

G1 set as sender with a interval of 15 for the range test plugin. Configured as a wifi client
G2 is this unit which is crashing and is what generated the log for this issue report.  Not configured for wifi.
G6 is the "receiver" which isn't as on for as long as G2 so as to not fill up the rangetest.csv (this  logging unit has been crashing but automatically reboots and continues logging but lets leave that for another day). Configured as a wifi client.

**Expected behavior**
Function as normal

**Device info:**
 - Device model: TBEAM T22_v1.1 20210222 915mhz
 - Software Version 1.2.50.41dcfdd
 - Radio config as below
 - `meshtastic --port COM5 --info
Connected to radio

Owner: Garry 2 (G2)
My info: { "myNodeNum": 2475426532, "hasGps": true, "numBands": 20, "firmwareVersion": "1.2.50.41dcfdd", "rebootCount": 34, "bitrate": 2418.3674, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 0.4033333 }
Nodes in mesh:  {'num': 2475426532, 'user': {'id': '!938c02e4', 'longName': 'Garry 2', 'shortName': 'G2', 'macaddr': '44:17:93:8c:02:e4', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': x, 'longitudeI': x, 'altitude': 65, 'batteryLevel': 100, 'time': 1641889048, 'locationSource': 'LOCSRC_GPS_INTERNAL', 'posTimestamp': 1641888934, 'altitudeHae': 63, 'altGeoidSep': -1, 'PDOP': 557, 'fixType': 3, 'satsInView': 4, 'latitude': x, 'longitude': x}, 'lastHeard': 1641889048}  {'num': 2475426076, 'user': {'id': '!938c011c', 'longName': 'Garry 1', 'shortName': 'G1', 'macaddr': '44:17:93:8c:01:1c', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': x, 'longitudeI': x, 'time': 1641888699, 'latitude': x, 'longitude': x}, 'lastHeard': 1641889047, 'snr': 5.0, 'lastReceived': {'from': 2475426076, 'to': 4294967295, 'id': 2050709698, 'rxTime': 1641889047, 'rxSnr': 5.0, 'hopLimit': 3, 'fromId': '!938c011c', 'toId': '^all'}, 'hopLimit': 3}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "ANZ", "positionBroadcastSmart": true }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw500Cr45Sf128", "psk": "AQ==", "name": "My Channel", "id": 1234 }

Primary channel URL: https://www.meshtastic.org/d/#ChYYASIBASoKTXkgQ2hhbm5lbFXSBAAA`

**Additional context**
Crash occurs halfway down the attached file. i then manually hit the reset switch which captured the bootup.

`
[crashdump3.txt](https://github.com/meshtastic/Meshtastic-device/files/7844774/crashdump3.txt)
`

mc-hamster, 2022-01-12:
Thanks for this. Could I ask you to reattach the log? It doesn't look like it took.

garthvh, 2022-01-26:
I would try 60 seconds for the range test plugin, 15 second intervals on virtually anything will flood the mesh.

closed 2022-01-26 by garthvh