meshtastic/firmware #1080: Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)

kingoffoxes, 2022-01-11:
**Describe the bug**
T-Beam stuck in reboot loop while rangetest logging. all set with shortfast

G1 set as sender with a interval of 15 for the range test plugin. Configured as a wifi client
G2 is this unit which is crashing and is what generated issue #1079. Not configured for wifi.
G6 is this unit its the "receiver" which isn't as on for as long as G2 so as to not fill up the rangetest.csv. Configured as a wifi client.

**To Reproduce**
Steps to reproduce the behavior:
meshtastic --port COM6 --set is_always_powered true

meshtastic --port COM6 --set range_test_plugin_enabled 1

meshtastic --port COM6 --set range_test_plugin_save 1

**Expected behavior**
Expect this to log without issues on shortfast setting. 
Reboot does not occur when the sending unit is off. 

**Device info:**
Device model: TBEAM T22_v1.1 20210222 915mhz
Software Version 1.2.50.41dcfdd

**Additional context**
radio config 

```
meshtastic --port COM6 --info
Connected to radio

Owner: Garry 6 (G6)
My info: { "myNodeNum": 2475420500, "hasGps": true, "numBands": 20, "firmwareVersion": "1.2.50.41dcfdd", "errorCount": 1, "rebootCount": 118, "bitrate": 2418.3674, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 2475420500, 'user': {'id': '!938beb54', 'longName': 'Garry 6', 'shortName': 'G6', 'macaddr': '44:17:93:8b:eb:54', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': x, 'longitudeI': x, 'altitude': 71, 'batteryLevel': 100, 'time': 1641892676, 'locationSource': 'LOCSRC_GPS_INTERNAL', 'posTimestamp': 1641892673, 'altitudeHae': 69, 'altGeoidSep': -1, 'PDOP': 388, 'fixType': 3, 'satsInView': 5, 'latitude': x, 'longitude': x}, 'lastHeard': 1641892676}  {'num': 2475426076, 'user': {'id': '!938c011c', 'longName': 'Garry 1', 'shortName': '?1C', 'macaddr': '44:17:93:8c:01:1c', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': x, 'longitudeI': x, 'time': 1641724655, 'latitude': x, 'longitude': x}, 'lastHeard': 1641891313, 'snr': 5.5}  {'num': 2475426532, 'user': {'id': '!938c02e4', 'longName': 'Garry 2', 'shortName': 'G2', 'macaddr': '44:17:93:8c:02:e4', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': x, 'longitudeI': x, 'time': 1641892322, 'latitude': x, 'longitude': x}, 'lastHeard': 1641892292, 'snr': 6.5, 'lastReceived': {'from': 2475426532, 'to': 4294967295, 'id': 1346189390, 'rxTime': 1641892292, 'rxSnr': 6.5, 'hopLimit': 3, 'fromId': '!938c02e4', 'toId': '^all'}, 'hopLimit': 3}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "ANZ", "positionBroadcastSmart": true, "gpsUpdateInterval": 5, "rangeTestPluginEnabled": true, "rangeTestPluginSave": true, "isAlwaysPowered": true }

Channels:
  PRIMARY psk=default { "txPower": 20, "modemConfig": "Bw500Cr45Sf128", "psk": "AQ==", "name": "My Channel", "bandwidth": 500, "spreadFactor": 7, "codingRate": 5, "id": 1234 }

Primary channel URL: https://www.meshtastic.org/d/#Ch8IFBgBIgEBKgpNeSBDaGFubmVsMPQDOAdABVXSBAAA

```

[crashdump5.txt](https://github.com/meshtastic/Meshtastic-device/files/7845180/crashdump5.txt)

mc-hamster, 2022-01-12:
Backtrace

jmcasler@MBP14inch2021 Downloads % ../Documents/GitHub/mc-hamster/Meshtastic-device/bin/exception_decoder.py -e firmware-tlora-v2-1-1.6-1.2.50.41dcfdd.elf backtrace.txt
stack:
0x400812f0: RadioLibInterface::isrLevel0Common(RadioLibInterface::PendingISR) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/RadioLibInterface.cpp:326
0x40081319: SX127x::clearDio0Action() at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tlora-v2-1-1.6/RadioLib/src/modules/SX127x/SX127x.cpp:1308
0x400813b5: __onPinInterrupt at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.c:274
0x400823fd: _xt_medint2_exit at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1262
0x400967c9: esp_rom_spiflash_program_page_internal at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0x40096802: esp_rom_spiflash_program_page_internal at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0x40096b59: esp_rom_spiflash_write_encrypted at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:516
0x40086ba5: spi_flash_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:373
0x40187ece: register_chipv7_phy at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7.c:1362
0x40191ce5: nvs::Page::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char) at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:327
0x401934cc: rijndaelEncrypt at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/crypto/aes-internal-enc.c:98
0x40195ab5: fast_hmac_sha256_vector at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/fast_crypto/fast_sha256.c:112
0x401964ae: esp_vfs_open at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:734
0x4019414d: hmac_sha1_vector at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/crypto/sha1.c:51
0x40191fca: nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:327
0x4019202f: nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:327
0x4019294c: nvs::PageManager::load(unsigned int, unsigned int) at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_pagemanager.cpp:32
0x40191839: nvs::Page::getVarDataTailroom() const at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_page.cpp:327
0x4019cc29: pm_set_sleep_type at ??:?
0x401ced36: btdm_controller_deinit at ??:?
0x4010c1fb: Print::printFloat(double, unsigned char) at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:234
0x400ea442: SPIClass::begin(signed char, signed char, signed char, signed char) at /home/runner/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:269
0x400ea509: SPIClass::setFrequency(unsigned int) at /home/runner/.platformio/packages/framework-arduinoespressif32/libraries/SPI/src/SPI.cpp:269
0x400de969: recordCriticalError(_CriticalErrorCode, unsigned int, char const*) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/NodeDB.cpp:572
0x400e2135: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::~basic_string() at /home/runner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:925
  \-> inlined by: registerHandlers(httpsserver::HTTPServer*, httpsserver::HTTPSServer*) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/http/ContentHandler.cpp:78
0x400e2205: registerHandlers(httpsserver::HTTPServer*, httpsserver::HTTPSServer*) at /home/runner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:925
0x400e2225: registerHandlers(httpsserver::HTTPServer*, httpsserver::HTTPSServer*) at /home/runner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/shared_ptr_base.h:925
0x400dda25: bool __gnu_cxx::__ops::_Iter_comp_iter<bool (*)(_MeshPacket const*, _MeshPacket const*)>::operator()<__gnu_cxx::__normal_iterator<_MeshPacket**, std::vector<_MeshPacket*, std::allocator<_MeshPacket*> > >, __gnu_cxx::__normal_iterator<_MeshPacket**, std::vector<_MeshPacket*, std::allocator<_MeshPacket*> > > >(__gnu_cxx::__normal_iterator<_MeshPacket**, std::vector<_MeshPacket*, std::allocator<_MeshPacket*> > >, __gnu_cxx::__normal_iterator<_MeshPacket**, std::vector<_MeshPacket*, std::allocator<_MeshPacket*> > >) at /home/runner/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/predefined_ops.h:123
[... cut ...]
0x400dda25: ReliableRouter::runOnce() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/DSRRouter.cpp:240
0x400d5352: concurrency::OSThread::run() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/concurrency/OSThread.cpp:45
0x400f3ddd: ThreadController::runOrDelay() at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/Thread/ThreadController.cpp:153
0x400dcdbc: loop() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/main.cpp:703
0x401181c1: loopTask(void*) at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19

mc-hamster, 2022-01-12:
Backtrace

jmcasler@MBP14inch2021 Downloads % ../Documents/GitHub/mc-hamster/Meshtastic-device/bin/exception_decoder.py -e firmware-tbeam-1.2.50.41dcfdd.elf backtrace.txt
stack:
0x400812f0: RadioLibInterface::isrLevel0Common(RadioLibInterface::PendingISR) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/RadioLibInterface.cpp:326
0x40081319: RadioLibInterface::isrRxLevel0() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/RadioLibInterface.cpp:326
0x400813b5: __onPinInterrupt at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-gpio.c:274
0x400823fd: _xt_lowint1 at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
0x400967c9: esp_rom_spiflash_read_status at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0x40096802: esp_rom_spiflash_wait_idle at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:413
0x40096b59: esp_rom_spiflash_erase_sector_internal at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:121
  \-> inlined by: esp_rom_spiflash_erase_sector at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/spi_flash_rom_patch.c:451
0x40086ba5: spi_flash_erase_range at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c:237
0x40187ece: esp_partition_erase_range at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c:294
0x40191ce5: spiffs_api_erase at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs_api.c:58
0x401934cc: spiffs_erase_block at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1164
0x40195ab5: spiffs_gc_erase_block at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c:15
0x401964ae: spiffs_gc_check at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_gc.c:168
0x4019414d: spiffs_object_append at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_nucleus.c:1219
0x40191fca: spiffs_hydro_write at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0x4019202f: spiffs_fflush_cache at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0x4019294c: SPIFFS_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/spiffs/src/spiffs_hydrogen.c:1121
0x40191839: vfs_spiffs_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/spiffs/esp_spiffs.c:496
0x4019cc29: esp_vfs_close at /home/kevinh/development/meshtastic/esp32-arduino-lib-builder/esp-idf/components/vfs/vfs.c:734
0x401ced36: VFSFileImpl::close() at /home/runner/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/vfs_api.h:31
0x4010c1fb: fs::File::close() at /home/runner/.platformio/packages/framework-arduinoespressif32/libraries/FS/src/FS.cpp:258
0x400ea442: RangeTestPluginRadio::appendFile(_MeshPacket const&) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/plugins/esp32/RangeTestPlugin.cpp:286
0x400ea509: RangeTestPluginRadio::handleReceived(_MeshPacket const&) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/plugins/esp32/RangeTestPlugin.cpp:139
0x400de969: MeshPlugin::callPlugins(_MeshPacket const&, RxSource) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/MeshPlugin.cpp:125
0x400e2135: Router::handleReceived(_MeshPacket*, RxSource) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/Router.cpp:180
0x400e2205: Router::perhapsHandleReceived(_MeshPacket*) at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/Router.cpp:180
0x400e2225: Router::runOnce() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/Router.cpp:180
0x400dda25: ReliableRouter::runOnce() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/mesh/DSRRouter.cpp:240
0x400d5352: concurrency::OSThread::run() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/concurrency/OSThread.cpp:45
0x400f3ddd: ThreadController::runOrDelay() at /home/runner/work/Meshtastic-device/Meshtastic-device/.pio/libdeps/tbeam/Thread/ThreadController.cpp:153
0x400dcdbc: loop() at /home/runner/work/Meshtastic-device/Meshtastic-device/src/main.cpp:703
0x401181c1: loopTask(void*) at /home/runner/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19

mc-hamster, 2022-01-12:
@kingoffoxes I suspect your filesystem has gotten corrupt. I've seen this once before. 

Could you try erasing your device's flash memory and then reinstalling meshtastic? LMK if this fixes your problem.

The hint here is 'rangiest.csv' is listed twice for the exact same filename:

??:??:?? 0 Filesystem files:
??:??:?? 0   /static/rangetest.csv
??:??:?? 0   /static/rangetest.csv
??:??:?? 0   /static/index.69df4442.js.gz

kokroo, 2022-01-12:
Guru Meditation error? What does it even mean?

kingoffoxes, 2022-01-12:
Thanks @mc-hamster i will try that now.

kingoffoxes, 2022-01-15:
Re-flashing solved the issue.

closed 2022-01-15 by kingoffoxes