meshtastic/firmware #1085: Save and reuse GPS almanac information

mc-hamster, 2022-01-13:
When the GPS gets a successful lock, save the received almanac information from the GPS satellites onto local flash.

On power on, load the saved almanac back to the GPS receiver.

Every day the GPS is still active, save the almanac again to flash.

The almanac is valid for 180 days.

runestone74, 2022-01-27:
How about uploading this to the device from a known good source, ie a connected mobile phone? This must speed up the first GPS lock I suspect? A timestamp from a known good timesource would also help a lot.

garthvh, 2022-09-11:
@caveman99 is this still relevant with all the GPS refactoring?

caveman99, 2022-09-27:
good question, i have never dump/reloaded the almanac and am not even sure this is possible with NMEA.

garthvh, 2022-09-28:
Closing this then, we have limited need for more GPS tasks, and it appears to be a part of the old UBLOX library we are not using any more.

closed 2022-09-28 by garthvh