meshtastic/firmware #1086: json/report is broken?

ANS-spb, 2022-01-13:
Airtime statistics in Meshtastic device JSON report (meshtastic.local/json/report) is maybe broken, I mean.

In the 1.2.49 firmware and earlier, airtime statistics were broken down into columns according to a period of 3600 seconds (1 hour). If there is no receive, after the expiration of the period, 0 was written in the rx_log column and a new column was considered. Now, if there is no receive, the count stops and a new column does not start.

Earlier:
```
"tx_log": [330304, 3235294, 19272, 34914, 3237478, 19272, 34914, 3239663, 19272, 34914, 3241847, 19272, 34914, 3244032, 19272, 34914, 3246216, 19272, 34913, 3248401, 19016, 34913, 3250585, 19016],
"rx_log": [0, 0, 0, 0, 0, 0, 41577, 0, 3252770, 19016, 0, 0, 34911, 3254955, 19016, 34911, 0, 0, 3257139, 19016, 34911, 3259323, 19016, 34911],
"rx_all_log": [0, 0, 0, 1233, 52793, 3270246, 19016, 34910, 3272431, 19016, 34910, 3274615, 19016, 34909, 3276800, 19016, 34909, 3278984, 19016, 34909, 3281169, 19016, 34909, 3283353],
```

Now:
```
"tx_log": [330304, 3235294, 19272, 34914, 3237478, 19272, 34914, 3239663, 19272, 34914, 3241847, 19272, 34914, 3244032, 19272, 34914, 3246216, 19272, 34913, 3248401, 19016, 34913, 3250585, 19016],
"rx_log": [41577, 3252770, 19016, 34911, 3254955, 19016, 34911, 3257139, 19016, 34911, 3259323, 19016, 34911, 3261508, 19016, 34911, 3263693, 19016, 34910, 3265877, 19016, 34910, 3268062, 19016],
"rx_all_log": [52793, 3270246, 19016, 34910, 3272431, 19016, 34910, 3274615, 19016, 34909, 3276800, 19016, 34909, 3278984, 19016, 34909, 3281169, 19016, 34909, 3283353, 19016, 34907, 3285538, 19016],

```

Also, after rebooting or reflashing device I was seen empty airtime statistics (tx_log, rx_log, rx_all_log columns was been filled by zeroes "0, 0, 0, 0, ". Now, after device starting, even after new flashing by device-install script, I see non-zero values in airtime report.

 - Device model: TBEAM 1.1. w/m8n
 - Software Version 1.2.50.41dcfdd

mc-hamster, 2022-01-13:
Thanks @ANS-spb. I'll look into it.

mc-hamster, 2022-01-13:
Confirmed. I can reproduce this.

mc-hamster, 2022-01-13:
@ANS-spb Just curious, what were you using this information for?

ANS-spb, 2022-01-13:
@mc-hamster I have an unstable connection with other nodes, using this information I evaluate the number of received packets from other devices on the mesh. Just diagnostic :)

mc-hamster, 2022-01-13:
Very good. Thanks.

Take a look at the channel utilization metric. That's more real time.

ANS-spb, 2022-01-14:
This is very interesting, thank you.
How should this value be interpreted? If there is no connection to other nodes, I see 0% channel usage. If I see 100%, is that bad too?
What is the optimal value?

mc-hamster, 2022-01-14:
That number is a rolling 1 minute window of all usage on the channel, from tx, rx and "noise" (valid lora packets that are not ours).

In my head, I think anything over 50% is high utilization.

I'm starting to configure some plugins not to transmit if utilization is above 25%.

ANS-spb, 2022-01-14:
Thanks for the explanation :) 
However, airtime stats in the JSON report gives convenient view of RH / TX throughout the day. I think, both metrics complement each other.

mc-hamster, 2022-01-14:
> Thanks for the explanation :)
> However, airtime stats in the JSON report gives convenient view of RH / TX throughout the day. I think, both metrics complement each other.

Yeap, just trying to give you an option until that's fixed again if you are on the newer firmware :)

ANS-spb, 2022-01-14:
> Yeap, just trying to give you an option until that's fixed again if you are on the newer firmware :)

Thank you :)

mc-hamster, 2022-01-15:
This will be addressed in the next build.

The bug was my bad. Thanks for reporting!

closed 2022-01-15 by mc-hamster

mc-hamster, 2022-01-16:
@ANS-spb 

If you'd like a new version with this fixed, go ahead and try out the bleeding build:

https://meshtastic.org/firmware