meshtastic/firmware #1089: Portduino + Waveshare SX1262-915M-LoRa-HAT

D4rk4, 2022-01-13:
I've tried to run meshtasticd_linux_amd64 1.2.50.41dcfdd, but got only SIMULATED radio.
Got message **Warning: Failed to find SX1262 radio**

dmesg:
`[ 9321.953801] usb 1-5: new full-speed USB device number 7 using xhci_hcd
[ 9322.081212] usb 1-5: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 9322.081213] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9322.081213] usb 1-5: Product: CP2102 USB to UART Bridge Controller
[ 9322.081214] usb 1-5: Manufacturer: Silicon Labs
[ 9322.081215] usb 1-5: SerialNumber: 0001
[ 9322.083134] cp210x 1-5:1.0: cp210x converter detected
[ 9322.084779] usb 1-5: cp210x converter now attached to ttyUSB0
`

meshtasticd:
`Portduino is starting, HWID=1, VFS root at /home/dark/.portduino/default
Setting up Meshtastic on Porduino...
writePin(fakeBusy, 32, 0)
��H??:??:?? 0 Filesystem files:
??:??:?? 0   /.
??:??:?? 0   /..
??:??:?? 0   /prefs
i2cbegin is not implemented
??:??:?? 0 No I2C devices found
??:??:?? 0 Meshtastic hwvendor=37, swver=1.2.50.41dcfdd, hwver=1.0
??:??:?? 0 Initial packet id 1804289383, numPacketId 4294967295
??:??:?? 0 Setting default channel and radio preferences!
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using Unset
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved preferences version 11
??:??:?? 0 Loading /prefs/radio.proto
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 3, using EU865
??:??:?? 0 region=3, NODENUM=0x4, dbsize=1
??:??:?? 0 Read RTC time as 1642081082 (cur millis 1) quality=0
??:??:?? 0 Using MSL altitude model
??:??:?? 0 Warning: No GPS found - running without GPS
??:??:?? 0 Starting meshradio init...
??:??:?? 0 (bw=250, sf=10, cr=4/7) packet symLen=4 ms, payloadSize=16, time 279 ms
??:??:?? 0 Set radio: name=MediumFast, config=5, ch=0, power=0
??:??:?? 0 Radio myRegion->freq: 865.200012
??:??:?? 0 Radio myRegion->spacing: 0.300000
??:??:?? 0 Radio myRegion->numChannels: 10
??:??:?? 0 Radio channel_num: 0
??:??:?? 0 Radio frequency: 865.200012
??:??:?? 0 Short packet time: 279 msec
??:??:?? 0 Set radio: final power level=22
setPinMode(fakeLoraIrq, 33, 0)
setPinMode(fakeLoraReset, 14, 1)
writePin(fakeLoraReset, 14, 0)
writePin(fakeLoraReset, 14, 1)
??:??:?? 3 SX126x init result -2
??:??:?? 3 Current limit set to 140.000000
??:??:?? 3 Current limit set result -2
??:??:?? 3 Warning: Failed to find SX1262 radio
??:??:?? 3 Starting meshradio init...
??:??:?? 3 (bw=250, sf=10, cr=4/7) packet symLen=4 ms, payloadSize=16, time 279 ms
??:??:?? 3 Set radio: name=MediumFast, config=5, ch=0, power=0
??:??:?? 3 Radio myRegion->freq: 865.200012
??:??:?? 3 Radio myRegion->spacing: 0.300000
??:??:?? 3 Radio myRegion->numChannels: 10
??:??:?? 3 Radio channel_num: 0
??:??:?? 3 Radio frequency: 865.200012
??:??:?? 3 Short packet time: 279 msec
??:??:?? 3 Using SIMULATED radio!
??:??:?? 3 API server listening on TCP port 4403
??:??:?? 3 (bw=250, sf=10, cr=4/7) packet symLen=4 ms, payloadSize=237, time 1545 ms
??:??:?? 3 myNodeInfo.bitrate = 153.398056 bytes / sec
??:??:?? 3 PowerFSM init, USB power=1
??:??:?? 3 Enter state: BOOT
??:??:?? 3 [mqtt] Connecting to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password large4cats
`

D4rk4, 2022-01-13:
Look's like the current realization work only with **CH341**, but Waveshare's HAT has **CP2102**.
https://github.com/meshtastic/Meshtastic-device/blob/2f7e200bef69ebe684f4af29c109c39541209eb8/src/portduino/PortduinoGlue.cpp#L70

closed 2022-02-04 by garthvh

D4rk4, 2022-02-04:
@garthvh why did you close this issue?

garthvh, 2022-02-04:
Pull requests are welcome, we do not support the waveshare hat currently