meshtastic/firmware #1093: Option to set public node

megabyte2010, 2022-01-15:
The android app could have an option to set the node as public. This option change channel settings to be open to public and maybe send location to some online map. If you want a closed network, disable that option.

mc-hamster, 2022-01-15:
What differentiates a node as public or closed?

megabyte2010, 2022-01-15:
You don't need to set and share channel name and settings. People that want to talk with another people in a city or beyond. It's a public channel. All nodes can connect with that channel and share their location in a map. Something like ham radio but for text.

mc-hamster, 2022-01-15:
That's the behavior of the default channel.

No need to set anything, it just works.

Is there anything else you're looking for?

kingoffoxes, 2022-01-17:
Perhaps the description or naming of the default channel can be made to be more clear that its public or that if you have configured your own channels that going back to the default channel is indeed certified as the "public channel"

garthvh, 2022-01-23:
Seems like we can close this and @kingoffoxes can send a pull request to the docs if there is some verbiage to adjust.

closed 2022-01-24 by garthvh