meshtastic/firmware #1100: Feature Request: MQTT Server Connection Status

Noki, 2022-01-16:
Currently there is no way to tell if the MQTT connection is established or not. We should have a status indicator variable that reflects the MQTT server connection status so it could be shown e.g. in the Meshtastic-web-app.

D4rk4, 2022-01-16:
Also in "My Info section" and LCD

garthvh, 2022-01-24:
I think this probably needs to be in the MQTT Client not the device firmware.

Noki, 2022-01-25:
> I think this probably needs to be in the MQTT Client not the device firmware.

The firmware needs to be adjusted to expose the status in protobuf and to display it on the LCD.

garthvh, 2022-01-25:
Plugins have their own screens, I am not sure we want to start putting optional plugin data on the main screen which seems to be what you are requesting.

Noki, 2022-01-25:
@garthvh 
I am/was not under the impression that MQTT is a plugin because it is not in the plugin folder. I also consider MQTT core functionality to close coverage gaps and not optional like some of the other plugins (e.g. environment sensors, store and forward, range test).

What I suggested is that we find a way to expose and show the connection status to the MQTT server since it is clearly missing and there is currently no way to see that a device is not connected. I think this is as fundamental as showing the Wifi SSID, the IP or the number of GPS satellites. When the MQTT server was down there was no way to diagnose this by looking at the screen or the web interface.

I really have no idea why you would consider such a request invalid.

garthvh, 2022-01-25:
You are right it is not a plugin, I still don't totally understand why we would want this in the Protobuf which puts the data over the mesh and on the main device screen. Why wouldn't the MQTT client software be responsible for status of MQTT?

Noki, 2022-01-25:
I don't care how the information ends up in the corresponding places but as far as I understand it all the data that is displayed in the web interface comes from requests where the Content-Type is `application/x-protobuf`.

[image]

mc-hamster, 2022-01-25:
I'm self assigning. I see where this can be added with low effort.

garthvh, 2022-01-25:
>  I think this is as fundamental as showing the Wifi SSID, the IP or the number of GPS satellites. 

I understand that for people using MQTT it is very important, but we are talking about generously 1% of users with MQTT enabled, GPS is probably at least 50% of devices. MQTT does not work reliably on the two devices I have it running on for testing. 

The protobuf is the device api and virtually everything in the protobuf is / can be sent over the mesh.

mc-hamster, 2022-01-25:
This can be added to MyNodeInfo which is only sent over the phone api. Also can be added to the json response in /json/report.

MyNodeInfo is starting to reach ble packet limits, but there'll be space for a simple bool.

garthvh, 2022-01-26:
That seems like a good place

Noki, 2022-08-25:
@mc-hamster Any update on this?

garthvh, 2022-09-03:
There is no code in the MQTT Module providing connection status, so we have no data to update this bool value with.

garthvh, 2022-09-03:
By my measurement a MyInfo packet is at 233 bytes right now as well the blue max is 255 and our header is 9-15 it seems

closed 2022-09-07 by garthvh