meshtastic/firmware #1103: Display TX indicator

mchisari, 2022-01-16:
This is a simple feature request. 
It would be nice to have. on display, a visual indication that the device is transmitting - such as "T" character or a dot in a corner, or some value like "@2/2" inverted.
This would give the user an hint on actual generated traffic and speed, and understand better what's going on the mesh.
However I don't know it it's easy to know from radio interface when TX begins and ends.

mc-hamster, 2022-01-16:
Currently there's the channel utilization metric, which is all tx and rx averaged over a minute, available on screen. Next week's version is going to have the TX which is just TX over an hour. That information is going to be available on the python API and to other apps as they implement.

Would that satisfy this request of know actual generated traffic and speed?

mchisari, 2022-01-17:
Actually I was thinking of an instantaneous indication - not an average.
Suppose you send an "hello" message from your phone; you could get an idea how long it takes before being transmitted, and how long to send it "on air"; if it seems too much you would keep your messages to minimum length. You'd also see when a message is acknowledged or relayed by your device.
I don't know if it's possible, it depends on the radio interface; maybe you send a block of data to radio, then it decides when to send it, so there's no way to precisely know at a certain time if it's transmitting or not?

closed 2022-01-26 by garthvh