meshtastic/firmware #1109: Device reboots on serial connect

sachaw, 2022-01-18:
Once a serial console is connected to a T-Beam for example, it will immediately reset.

It appears that it may be implementation specific (i.e. happens with some clients and not others)

https://www.microsoft.com/en-us/p/serial-debug-assistant/9nblggh43hdm#activetab=pivot:overviewtab for example always resets.

mkinney, 2022-01-18:
Can you have add the `--support` option from the meshtastic (python) cli?

mjrsnyder, 2022-02-21:
I'm seeing this as well on `1.2.55` on two different [Lora V2.1-1.6](https://meshtastic.org/docs/hardware/supported/lora#lora-v21-16) boards.

```
(meshtastic-env) PS C:\Users\mjrsn> meshtastic --info
Traceback (most recent call last):
  File "C:\Program Files (x86)\Python38-32\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files (x86)\Python38-32\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\mjrsn\meshtastic-env\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "c:\users\mjrsn\meshtastic-env\lib\site-packages\meshtastic\__main__.py", line 912, in main
    common()
  File "c:\users\mjrsn\meshtastic-env\lib\site-packages\meshtastic\__main__.py", line 694, in common
    client = meshtastic.serial_interface.SerialInterface(args.port, debugOut=logfile, noProto=args.noproto)
  File "c:\users\mjrsn\meshtastic-env\lib\site-packages\meshtastic\serial_interface.py", line 57, in __init__
    StreamInterface.__init__(self, debugOut=debugOut, noProto=noProto, connectNow=connectNow)
  File "c:\users\mjrsn\meshtastic-env\lib\site-packages\meshtastic\stream_interface.py", line 50, in __init__
    self.connect()
  File "c:\users\mjrsn\meshtastic-env\lib\site-packages\meshtastic\stream_interface.py", line 74, in connect
    self._waitConnected()
  File "c:\users\mjrsn\meshtastic-env\lib\site-packages\meshtastic\mesh_interface.py", line 401, in _waitConnected
    raise Exception("Timed out waiting for connection completion")
Exception: Timed out waiting for connection completion
```
```
(meshtastic-env) PS C:\Users\mjrsn> meshtastic --support

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10-10.0.19041-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.2.85
 Executable: C:\Users\mjrsn\meshtastic-env\Scripts\meshtastic
 Python: 3.8.6 CPython MSC v.1927 32 bit (Intel)

Please add the output from the command: meshtastic --info
```

garthvh, 2022-02-21:
Windows?

mjrsnyder, 2022-02-22:
> Windows?

Yes, this doesn't occur using the same device on a Linux host.

karamo, 2022-03-04:
The T-Beam V1.1 (T33_v1.1 with GPS) and the TTGO LoRa T3_V1.6.1 have different reactions by RTS/DSR via the USB.

garthvh, 2022-03-04:
> The T-Beam V1.1 (T33_v1.1 with GPS) and the TTGO LoRa T3_V1.6.1 have different reactions by RTS/DSR via the USB.

Using windows?

karamo, 2022-03-04:
Yes, under Windows 10 64Bit.
In some cases the first use of Meshtastic did a hard reset/reboot, the following uses had no such effect; on T33_v1.1

karamo, 2022-03-04:
Sorry, 2nd time wrong: I meant a T22_v1.1 !!!

caveman99, 2022-09-24:
haven't seen this in a long time, please reopen if still a problem with recent 1.3

closed 2022-09-24 by caveman99