meshtastic/firmware #1111: Feature Proposal: adding timestamps to GPS positions

mchisari, 2022-01-18:
I believe that in networks like Meshtastic, giving no information is better than a false one. 

GPS data can be old, because you've heard the remote node long ago, or because it is currently unable to get fix - maybe it relies on phone GPS because it hasn't one on its own. To avoid mistakes old GPS data could be discarded, but I think there's a better way to handle this.
A device could send an update saying “half an hour ago I was at…” which is better that than saying “I believe I’m at…” or  “I don’t know where I am now”.

So, I propose that:
- When a device shares its position, it adds a timestamp stating when that position was taken
- Every device receiving it should store GPS timestamp with coordinates, along with a "received" timestamp
- This information "xxx minutes ago", relative to GPS timestamp, should be displayed by device with direction and distance of the other node; should also be shown on Android map of other users.
- The "received" timestamp should instead be shown in node list, both for device and Android app, so as to know when it was last heard.
At the moment the map and node “time” can be different because map uses time from position packet and node from nodeinfo packet. The logic instead should be: for each node store GPS position with its timestamp, and store last heard timestamp regardless of the type of received packet.
This would be more intuitive and informational.

garthvh, 2022-01-18:
These fields are already on position https://meshtastic.org/docs/developers/protobufs/api#position

andrekir, 2022-01-19:
> - When a device shares its position, it adds a timestamp stating when that position was taken
> - Every device receiving it should store GPS timestamp with coordinates, along with a "received" timestamp
> - The "received" timestamp should instead be shown in node list, both for device and Android app, so as to know when it was last heard.
> - At the moment the map and node “time” can be different because map uses time from position packet and node from nodeinfo packet. The logic instead should be: for each node store GPS position with its timestamp, and store last heard timestamp regardless of the type of received packet.

correct. this is all current behavior. `position.timestamp` is stored for every node's last position packet, `nodelist.lastheard` is updated for any activity/packet.

> - This information "xxx minutes ago", relative to GPS timestamp, should be displayed by device with direction and distance of the other node; should also be shown on Android map of other users.

the problem of switching the node page to `position.timestamp` is missing `nodeinfo.lastheard`.

`position.timestamp` is relevant for direction and distance, but screen space is limited for both. unless the format is changed to a more compact form-factor (like "30s", "13m", "32 days", etc) on each left and right corners.

on the android app you can check both values. `position.timestamp` is what is displayed on mapview and on the nodes list tab `nodelist.lastheard`.

something I believe is missing from the device GPS page (with time, coordinates, etc), is showing the same "xxx minutes ago" for when our last position was sent (from `position.timestamp`).

sorry if I wasn't clear and hope it makes more sense.

mchisari, 2022-01-19:
Thanks for your patience... I just elaborate a bit, as a simple new user. 

The logic is as expected, and compact time is both concise and convenient, nothing to complain. 
I've noticed that Android app most of the times shows on the map mine and other nodes position with "now" indication; at the same time, on user list, I see other node was received several hours ago. 
I'm confused. I'd be surprised to see nodes on the map ticking "10s... 11s..." - I guess the labels are static; so it could happen that they look ahead of last heard time, which is refreshed faster; but aren't 4+ hours a bit too much? 
I guess the map is generated just after getting GPS position, so it displays "now" and thus remains; but why "now" if node position was received hours ago? Suggestion: if regenerating the map is expensive, I'd do a little effort to add h:m/d:h/m:d label, eventually switching to short user name. 
As to device, I notice that other station distance/direction is lost as soon as the phone goes to sleep. Not a big issue actually - if it was my primary goal, I'd have bought a device with a GPS; it just adds uncertainty to app behavior...

andrekir, 2022-01-20:
> I've noticed that Android app most of the times shows on the map mine and other nodes position with "now" indication; at the same time, on user list, I see other node was received several hours ago. [...]

you're right. `position.timestamp` shouldn't be newer than `nodeinfo.lastheard`.

I'm seeing the same behavior on my devices. setup a new pair of test devices and... their position is not even showing on the map. strange.

I'll have this issue moved to Android and add it to my queue, thanks for reporting!