meshtastic/firmware #1112: No GPS on TBEAM0.7

notthetup, 2022-01-19:
Please - if you just have a question (i.e. not a bug report or a feature proposal), post in our [forum](https://meshtastic.discourse.group/) instead.

**Describe the bug**
I flashed the latest beta (also tried alpha) release on a Tbeam0.7 and I see `No GPS` on the screen even after the GPS PPS LED starts blinking..

**Device info:**
 - Device model: TBEAMv0.7
 - Software Version : firmware-tbeam0.7-1.2.52.b63802c.bin

This is the command I used for flashing..

```
./device-install.sh firmware-tbeam0.7-1.2.52.b63802c.bin
```

Flashing log seemed fine.

```
Trying to flash firmware-tbeam0.7-1.2.52.b63802c.bin, but first erasing and writing system information
esptool.py v3.2
Found 2 serial ports
Serial port /dev/cu.usbserial-02066329
Connecting.....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:9e:bd:f0:bd:3c
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 4.9s
Hard resetting via RTS pin...
esptool.py v3.2
Found 2 serial ports
Serial port /dev/cu.usbserial-02066329
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:9e:bd:f0:bd:3c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x0000ffff...
Compressed 61440 bytes to 11950...
Wrote 61440 bytes (11950 compressed) at 0x00001000 in 1.2 seconds (effective 397.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
esptool.py v3.2
Found 2 serial ports
Serial port /dev/cu.usbserial-02066329
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:9e:bd:f0:bd:3c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00390000 to 0x003fffff...
Compressed 458752 bytes to 261713...
Wrote 458752 bytes (261713 compressed) at 0x00390000 in 23.2 seconds (effective 158.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
esptool.py v3.2
Found 2 serial ports
Serial port /dev/cu.usbserial-02066329
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 7c:9e:bd:f0:bd:3c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00010000 to 0x001affff...
Compressed 1703472 bytes to 990357...
Wrote 1703472 bytes (990357 compressed) at 0x00010000 in 87.3 seconds (effective 156.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

This is the seriallog when rebooting the device. Note the `??:??:?? 2 ERROR: No UBLOX GPS found`

```
//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=reset
??:??:?? 0 Setting forceSoftAP = 1
??:??:?? 0 Filesystem files:
??:??:?? 0   /static/android-512.png.gz
??:??:?? 0   /static/Logo_Black.svg.gz
??:??:?? 0   /static/sw.js.gz
??:??:?? 0   /static/robots.txt.gz
??:??:?? 0   /static/index.14c9dc75.js.gz
??:??:?? 0   /static/android-192.png.gz
??:??:?? 0   /static/favicon-32x32.png.gz
??:??:?? 0   /prefs/db.proto
??:??:?? 0   /static/vendor.07b56cc4.js.gz
??:??:?? 0   /prefs/radio.proto
??:??:?? 0   /static/Logo_White.svg.gz
??:??:?? 0   /prefs/channels.proto
??:??:?? 0   /static/index.html.gz
??:??:?? 0   /static/favicon.ico.gz
??:??:?? 0   /static/site.webmanifest.gz
??:??:?? 0   /static/workbox-b7e829be.js.gz
??:??:?? 0   /static/manifest.webmanifest.gz
??:??:?? 0   /static/touch-icon.png.gz
??:??:?? 0   /static/safari-tab.svg.gz
??:??:?? 0   /static/index.b690784f.css.gz
??:??:?? 0   /static/.gitkeep
??:??:?? 0   /static/favicon-16x16.png.gz
??:??:?? 0 I2C device found at address 0x34
??:??:?? 0 I2C device found at address 0x3c
??:??:?? 0 ssd1306 display found
??:??:?? 0 done
??:??:?? 0 Meshtastic hwvendor=6, swver=1.2.52.b63802c, hwver=1.0
??:??:?? 0 Setting random seed 3788242051
??:??:?? 0 Total heap: 249776
??:??:?? 0 Free heap: 215564
??:??:?? 0 Total PSRAM: 4194252
??:??:?? 0 Free PSRAM: 4194252
??:??:?? 0 NVS: UsedEntries 234, FreeEntries 396, AllEntries 630, NameSpaces 6
??:??:?? 0 Setup Preferences in Flash Storage
??:??:?? 0 Number of Device Reboots: 2
??:??:?? 0 Initial packet id 799462479, numPacketId 4294967295
??:??:?? 0 Setting default channel and radio preferences!
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using Unset
??:??:?? 0 Loading /prefs/db.proto
??:??:?? 0 Loaded saved preferences version 11
??:??:?? 0 Loading /prefs/radio.proto
??:??:?? 0 Loading /prefs/channels.proto
??:??:?? 0 Number of Device Reboots: 2
??:??:?? 0 Expanding short PSK #1
??:??:?? 0 Wanted region 0, using Unset
??:??:?? 0 region=0, NODENUM=0xbdf0bd3c, dbsize=1
??:??:?? 0 Using analog input 35 for battery level
??:??:?? 0 Read RTC time as 0 (cur millis 244) quality=0
??:??:?? 0 Using MSL altitude model
??:??:?? 0 WANT GPS=1
??:??:?? 0 Setting GPS power=1
??:??:?? 2 ERROR: No UBLOX GPS found
??:??:?? 2 Hoping that NMEA might work
??:??:?? 2 WANT GPS=1
??:??:?? 2 GxGSA NOT available
??:??:?? 2 External Notification Plugin Disabled
??:??:?? 2 Turning on screen
??:??:?? 2 Starting meshradio init...
??:??:?? 2 (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=16, time 2269 ms
??:??:?? 2 Set radio: name=LongSlow, config=3, ch=0, power=0
??:??:?? 2 Radio myRegion->freq: 903.080017
??:??:?? 2 Radio myRegion->spacing: 2.160000
??:??:?? 2 Radio myRegion->numChannels: 13
??:??:?? 2 Radio channel_num: 0
??:??:?? 2 Radio frequency: 903.080017
??:??:?? 2 Short packet time: 2269 msec
??:??:?? 2 Set radio: final power level=17
??:??:?? 2 Current limit set to 100.000000
??:??:?? 2 Current limit set result 0
??:??:?? 2 RF95 init result -19
??:??:?? 2 Current limit set to 100.000000
??:??:?? 2 Current limit set result 0
??:??:?? 2 RF95 Radio init succeeded, using RF95 radio
??:??:?? 2 WiFi ... Forced AP Mode
??:??:?? 2 Waiting for SSL Cert to be generated.
??:??:?? 2 Existing SSL Certificate found!
??:??:?? 2 Retrieved Private Key: 1192 Bytes
??:??:?? 2 Retrieved Certificate: 780 Bytes
??:??:?? 2 SSL Cert Ready!
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
??:??:?? 2 ************ [WiFi-event] event: 0 ************
??:??:?? 2 WiFi interface ready
??:??:?? 2 Starti[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
ng (Forced) WIFI AP: ssid=meshtasticAdmin, ok=1
??:??:?? 2 mDNS responder started
??:??:?? 2 mDNS Host: Meshtastic.local
??:??:?? 2 MY IP AP ADDRESS: 192.168.42.1
??:??:?? 2 Initializing Web Server ...
??:??:?? 2 Starting Secure Web Server...
??:??:?? 2 Starting Insecure Web Server...
??:??:?? 2 Web Servers Ready! :-)
??:??:?? 2 API server listening on TCP port 4403
??:??:?? 2 (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=237, time 13869 ms
??:??:?? 2 myNodeInfo.bitrate = 17.088470 bytes / sec
??:??:?? 2 PowerFSM init, USB power=1
??:??:?? 2 Enter state: BOOT
??:??:?? 2 Setting CPU to 240mhz because WiFi is in use.
Connected to radio
```

This is the device.

[image]

notthetup, 2022-01-19:
Yikes! I realised that I have TBEAM v1.x instead of a v0.7. 

I went through the list of TBEAMs here https://meshtastic.org/docs/hardware/supported/tbeam to find the one which looks like mine, and got confused.

Sorry!

closed 2022-01-19 by notthetup

caveman99, 2022-01-19:
the 3 pushbuttons are a dead giveaway for a 'tbeam'. The 'tbeam 0.7' has a mechanical switch near the USB port.