meshtastic/firmware #1114: platformio.ini  missing default for tlora-v2-1-1.6 board

costonisp, 2022-01-19:
While trying to compile for TTGO  T3 V1.6.1  I had some difficulties when I choose 
default_envs = tlora-v2

So added  default_envs = tlora-v2-1-1.6  as line 20 in platformio.ini

thebentern, 2022-01-19:
That board is actually the tlora v2-1-1.6. TTGO's labeling is unfortunately particularly misleading on that board. I have a few of them. 

Also on the the default_envs, those are only a convenience to uncomment so that you don't need to specify the target every time. You can instead just run `platformio run --environment tlora-v2-1-1.6 --target upload --upload-port yourcomport`.
You can also run this from the VSCode extension if you have that:
[image]

costonisp, 2022-01-20:
tnx for your explanation.
I am a atom/platformio beginner and it took me some time to find out I had to use tlora-v2-1-1.6 as default, otherwise a added user_button does not work.
If it had been at the top in platformio.ini I would have uncommented it right away, but there was no entry for this board.

So my suggestion to add it as line 20 in platformio.ini . That makes it eaysier for beginners without digging in the source code to find this board.

I have more 'difficulties' with this board, the voltege measurement is inaccurate but I opened a topic for this in the forum.

thebentern, 2022-01-20:
Added default_env. Closing

thebentern, 2022-01-20:
@costonisp if you're confident in that ADC change for battery level, would you mind putting in a pull request with those changes?
This is the file that contains the config for that particular board now (this changed very recently):
[image]

closed 2022-01-20 by thebentern