meshtastic/firmware #1125: Add a user defineable variable to compensate for battery display.

costonisp, 2022-01-22:
Hardware details can change and it looks like the voltage dividers , even for the same model , can vary from device to device,
So the best option i.m.h.o. would be a user definable compensation variable. In the web interface that can finetune the ADC_MULTIPLIER in percent-points.
For the CLI it may be a command like meshtastic --set battery_compensation nn
Where the user can calculated this factor by measuring the battery divided by the reading on the display * 100 or
when user has no mulimeter he must charge the battery for 24h or so and assume it is 4.17 Volt and then read the display voltage and calculate the factor.

garthvh, 2022-01-22:
Look at the variant for the RAK devices, there is now a Heltec variant where you can do this work if desired.  I would not anticipate using the CLI or a client app to set this value based on how the code works.

https://github.com/meshtastic/Meshtastic-device/blob/master/variants/heltec_v2/variant.h

costonisp, 2022-01-23:
One compensation-value per variant  would  be OK if we are sure that they would fit all the devices of that variant.  But I fear that is not true, at least for the heltec2.1.  I own 2 identical devices and the compensation value is different for both. One device needs 3.54 and the other device needs 3.69.
Also in this post on the forum: meshtastic.discourse.group/t/heltec-esp32-only-charging-to-40-at-3-65v/4515 another user reported he needed  3.75 as compensation value.

This cannot be solved with ONE simple value in the variant section.
I do not know if other devices have the same variation from device to device maybe the heltec2.1 is a bad example.

garthvh, 2022-01-23:
There is an approach for the RAK boards on the discourse thread that is probably the place to start now that there is a variant file for the Heltec.

garthvh, 2022-01-23:
[Duplicate of](https://github.com/meshtastic/Meshtastic-device/issues/769)

costonisp, 2022-01-23:
Indeed there are more serious things in the code than this issue.
Maybe someone will pick it up again in the future.
Will close it for now.

Have edited the variant.h for the heltecv2
have set the multiplier for a arbitrary value of 3.6 which will do better than the 3.2

closed 2022-01-23 by costonisp