meshtastic/firmware #1126: Bug: T-Echo - When transmitting on LoRa "soft button" is pressed

ernax78, 2022-01-22:
When the radio is transmitting, it seems like the soft button is being pressed for some reason.

1. This spams the log (debug)
2. This might cause undesired operation should "long press" be connected to anything in the future to the soft button.

How to reproduce:

1. Hook up to serial
2. Wait for radio to do a transmission

Suggested action / work around:

Uncomment "Long press" for touch debug code
Write a warning in the "Long press start / stop / long press" functions about this issue so it's investigated before hooking this up.

Thanks to @caveman99  for pointing this out.

```
06:07:02 212584 [RadioIf] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=18, time 2367 ms
06:07:02 212584 [RadioIf] AirTime - Packet transmitted : 2367ms
06:07:03 212585 [Button] touch long press start!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212585 [Button] touch press long!
06:07:03 212586 [Button] touch press long!
06:07:03 212586 [Button] touch press long!
06:07:03 212586 [Button] touch press long!
06:07:03 212586 [Button] touch press long!
06:07:03 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [Button] touch press long!
06:07:04 212586 [RadioIf] Completed sending (id=0x28bc53a4 Fr0x14 To0xff, WantAck0, HopLim1 Ch0x77 encrypted rxtime=944201219 rxSNR=6.75 priority=64)
06:07:04 212587 [Button] touch long press stop!
06:07:19 212602 [Power] Battery: usbPower=1, isCharging=1, batMv=4873, batPct=100
```

caveman99, 2022-01-22:
The tiny piece of metal under the (capacitative) soft button simply picks up the HF from the transmisson and thinks it's a finger in the vicinity. A simple solution would be to disable the button during transmission, which does not occur that frequently.

mc-hamster, 2022-01-23:
I have not looked in our implementation, but the ADC values for the threshold of 'touch' vs 'not touched' will just need to be tweaked to below the threshold that is being triggered for this situation.

kokroo, 2022-01-28:
> When the radio is transmitting, it seems like the soft button is being pressed for some reason.
>
> 1. This spams the log (debug)
> 2. This might cause undesired operation should "long press" be [...]

Hey, I am trying to fix T-Echo's firmware. Could you help me out on the forums or Discord by testing a few things? Thanks!

commit 1e4779eb (2022-03-29): Merge pull request #1334 from meshtastic/patch-1126

fixes #1126
commit 561a01ec (2022-03-29): fixes #1126

Since i found no easy way to change the button threshold i removed the unused stubs and their debug messages. They can be added back in once they're used for something.
commit 3b8455ff (2022-03-29): Merge pull request #1335 from meshtastic/patch-1176

Fix #1126
closed 2022-03-29 by caveman99