meshtastic/firmware #1137: Home Assistant

joshpirihi, 2022-01-24:
It'd be really cool to have a proper Home Assistant integration that allows for 2-way interaction between your home assistant instance and the mesh.

_commence brain dump_

The integration should:
- register node positions within home assistant
- allow messages to be sent to the mesh (broadcast or direct).  Eg from an automation (garage door opens -> message sent to my meshtastic node)
- handle incoming messages in some way (so you could message
- connect to the mesh via mqtt (to a gateway node mounted remotely and connected to mqtt), or via direct usb connection to a node
- collect environmental plugin data, and publish as sensors within home assistant
- allow control of the remote GPIO pins of nodes _this would be so awesome_
- allow communication via the Serial Plugin
- link to the gateway node's webUI, if applicable

The integration should not:
- provide any wifi/network extension.  Meshtastic does not have the bandwidth for this

crossan007, 2022-01-24:
The Meshtastic protobuf defines a number of application port; with the idea that more may be added: https://github.com/meshtastic/Meshtastic-protobufs/blob/master/portnums.proto#L9

I am thinking about registering ESPHome as an application port, so that other "real Meshtastic" radios can still forward ("mesh") ESPHome packets

joshpirihi, 2022-01-24:
any idea what sort of bandwidth ESPHome uses?  The issue would be by creating a wifi network, devices will generally expect more than a few bytes per second and I can see the mesh being swamped very very easily.

crossan007, 2022-01-24:
Not totally sure, but it's all sensor data (I think wrapped in protobufs).

joshpirihi, 2022-01-24:
protobufs you say?  an ESPHome application port number would be perfect

pete1450, 2022-01-26:
I've been looking all over for an option that in my head was titled "Meshtastic... but with any sensors". Meshtastic+ESPHome would just be pure gold. Opens up a ton of possibilities. Exactly what I've been imagining to get set up across my parent's fruit orchards. I'm open to chipping in if I see somewhere I can help.

mc-hamster, 2022-01-26:
> I've been looking all over for an option that in my head was titled "Meshtastic... but with any sensors". Meshtastic+ESPHome would just be pure gold. Opens up a ton of possibilities. Exactly what [...]

Get a hold of @crossan007 on our Discord. He's discussed an integration with ESPHome.

joshpirihi, 2022-01-26:
It sounds like a module for ESPHome that allows communications over some RF link (not wifi) is the sort of thing you're after.  You could certainly write a module that's compatible with the meshtastic protocol, but at that point there doesn't seem much point in making it compatible - other meshtastic nodes won't be able to interact other than forwarding traffic.  

The idea I was more leaning to is an integration in HA that gives HA visibility into and messaging access to the mesh, running on meshtastic nodes.  In that instance you could certainly pull meshtastic temperature sensor data into HA, process somehow, and command meshtastic GPIO pins from HA.  Would that go some way to solving your use case?  What sort of sensors are you wanting to integrate here?

That said, I know very very little about ESPHome, and don't know if it's possible or feasible to add another transport layer to it.  It could well be very easy!

garthvh, 2022-01-27:
Seems to me like anything that is restricted to ESP only devices and really focused on WiFi not a great choice, the NRF52 chip is so much lower power it is a considerably better remote sensor node.

closed 2022-09-11 by sachaw