meshtastic/firmware #1143: Feature Proposal / Request: Adafruit MCP9808 & INA260 (Environmental Measurement Plugin)

RicInNewMexico, 2022-01-26:
**Feature Request**
Would love someone that's more code-oriented to take a look at adding support for adafruit's MCP9808 i2c temperature breakout and their INA260 i2c voltage / current / power sensor breakout. 

**Product Links**
[MCP9808](https://www.adafruit.com/product/1782?gclid=CjwKCAiA3L6PBhBvEiwAINlJ9HSB6glezZUUruR-OiYowAhuuLZab39ea2_7BVtRgbdJ30NJ_nMyxRoCJ7cQAvD_BwE)
[INA260](https://www.adafruit.com/product/4226?gclid=CjwKCAiA3L6PBhBvEiwAINlJ9D9tf1Hxu5DGdJSzVjWfskfvsOyGFseyu-X-aT7_Fj2mehLw5mcfJhoCtbAQAvD_BwE)

**Reasoning**
The MCP9808 is a MUCH lower cost (<$5.00 USD) solution for temperature monitoring where other environmental elements like barometric pressure, etc are not needed and opens up the possibility of separately monitoring battery pack temp, enclosure temp, outside temp, etc. 

The INA260 is a breakout board capable of measuring voltage (up to 36VDC) as well as current (up to 15A). Adding support for the INA260 would allow those of us using higher voltage solar panels (12v panels for example) the ability to monitor the supplied voltage and current draw of the solar panel during daytime charging. 

**Other Notes**
As both are i2c devices the implementation should be very similar to already supported i2c devices.

caveman99, 2022-01-26:
If I2C stuff is working in principle, we might put in an abstraction layer like the input plugin. The sensor devices work similar, define I2C addresses and a formula to convert raw readings into something aligned to the SI system.

garthvh, 2022-01-26:
@mkinney do you have a process for adding new sensors yet?

mkinney, 2022-01-26:
Not really. I'm still getting up to speed on the code, so I'm not sure I can do that part by myself yet.

If someone has some new/different sensors and is willing to test, then we need to buy a few for testing/development.

If someone wants to add the sensors to the codebase themselves, and they do their own testing, they can just submit a PR.

@RicInNewMexico Do you have these sensors or are you just "window shopping"? I do like to support AdaFruit/LadaAda.

garthvh, 2022-01-27:
After some discussion it seems like the place to start is to use I2C and sensors with Adafruit libraries. @thebentern is working on a common interface for sensors so they are relatively easy to add.  We also need to be considerate of space, especially on esp32 devices with the WebUI where space is now at a premium.  Long term it may not be possible to run the web interface and all of the sensors off of the same device.

garthvh, 2022-01-27:
@RicInNewMexico we will add these two sensors to our list.  May take a little while as I don't think we have these two ordered yet.

RicInNewMexico, 2022-01-27:
> @RicInNewMexico Do you have these sensors or are you just "window shopping"? I do like to support AdaFruit/LadaAda.

Sorry for the delay, had a long day at work. 

I don't have them in-hand yet, but I've already placed an order with them on it. Just waiting on them to come in. So it'll be a little while anyway. Once they arrive I'd be more than happy to provide remote access to a device with the sensors attached if it would be of any use to the devs.

RicInNewMexico, 2022-02-04:
Just an update for @garthvh et al, 

Still waiting on my sensors from adafruit, USPS shows the package still 3 days out (so much for priority mail).. on the upside, I have all of the code finished for the MCP9808, just waiting for them to arrive to do the final testing and submit a PR. Guess this could have been assigned to myself, lol..

garthvh, 2022-02-14:
@RicInNewMexico are both sensors supported now or are we only half way?

RicInNewMexico, 2022-02-15:
> @RicInNewMexico are both sensors supported now or are we only half way?

We're halfway there. The MCP9808 is now supported, but I haven't gotten around to writing the code to support the INA260. You can close this if you want, it's on my to-do list.. I just got sidetracked by re-writing the MCP9808 implementation so it didn't rely on the Adafruit MCP9808 library (the library isn't **really** needed and added 1.8kb to the firmware that could be avoided).

I'll be doing a new PR here in the next couple days or so to submit a replacement for the code previously merged under PR #1188. I just want to finish testing it first.

garthvh, 2022-02-15:
Just checking in on progress, I saw the MCP9808 pulls.  @mkinney has a INA219 from rak so maybe we can do those as a family of power sensors.

closed 2022-06-22 by thebentern