meshtastic/firmware #1151: Feature proposal: Simple "ping" response configurable with on/off setting

runestone74, 2022-01-27:
I think it would be nice to have a "ping" option that would ie answer "pong" when message received.

This would have many benefits, like testing that the device is alive and/or in radio range.

It could be on/off configurable to avoid unintended replys from devices that shouldn't answer on "ping" requests.

This could also be implemented directly in to the Meshtastic android/ios app, just as an auto-reply for the "ping" command.

garthvh, 2022-01-29:
We have a number of tools for this kind of testing already, range testing plugin could work and both phone apps have extensive logging of mesh activity. 

Pull requests are welcome, closing for now.

closed 2022-01-29 by garthvh

mc-hamster, 2022-01-29:
> I think it would be nice to have a "ping" option that would ie answer "pong" when message received.
>
> This would have many benefits, like testing that the device is alive and/or in radio range.
>
> It [...]

This is already built into the device code and python. See:

https://meshtastic.org/docs/software/python/python-cli#--sendping

VilemR, 2025-02-20:
Ping feature can be easilz implemented in custom firmware - see https://meshtastic-czbrno.blogspot.com/2025/02/meshtastic-kompilace-vlastniho-firmware.html and translate it from Cyech to English using google. It is done using new module - available here 
https://github.com/VilemR/meshtstic_modules_mod