meshtastic/firmware #1161: Only use Sf(7-12) no longer use Sf(512-4096) chips/symbol

costonisp, 2022-01-27:
It is confusing using spread-factor SF as simpel number and in other places use the chips/symbol which is just the number 2 to the power of Sf  (2^Sf)
For instance the modem_settings consequently uses the c/s setting.

So when in the cli do:  meshtastic --ch-set spread_factor 11 --ch-set coding_rate 5 --ch-set bandwidth 62  --ch-index 0

 meshtastic --info gives:
PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==", "bandwidth": 62, "spreadFactor": 11, "codingRate": 5 }

Here we see in one line Sf4096 and spreadFactor11 which is confusing

Maybe this example is an issue for the Meshtastic-python repository, but I see it  also in the device software..

When in all the repositories (where it is used) the use of chips/symbol is replaced by the Sf(7-12) i think meshtastic would become more transparent to the user.

mc-hamster, 2022-01-27:
good callout

mc-hamster, 2022-02-14:
Closing. This will be done in 1.3

closed 2022-02-14 by mc-hamster