meshtastic/firmware #1164: Feature proposal: iOS app to use SiriKit for messages

runestone74, 2022-01-28:
https://developer.apple.com/documentation/sirikit/messaging

This will enable handsfree interaction with the Meshtastic device, especially useful via CarPlay or when riding a bike.

runestone74, 2022-01-28:
Might as well go for full CarPlay functionality :) https://developer.apple.com/carplay/documentation/CarPlay-App-Programming-Guide.pdf
"Messaging must be handled using SiriKit. Your app must support sending a message,
searching for messages, and setting attributes on a message."

garthvh, 2022-01-28:
An actual WatchOS app beyond notification, CarPlay and TVOS are way in the backlog as they all require maintaining a parallel neutered version of the BLE code.  iOS, iPadOS and macOS are all in beta now.

runestone74, 2022-01-28:
I understand, but it will go a long way just to be able to use Siri
https://developer.apple.com/forums/thread/692457

garthvh, 2022-01-28:
You can dictate to Siri from the keyboard.  The app does not send sms / iMessages which is required to use INSendMessageIntent, this is not an option for the Meshtastic app currently.