meshtastic/firmware #1168: Feature Proposal: Flag nodes as stolen

LolloDev5123, 2022-01-30:
A proto/function that allows a node (from webpanel, android app, etc.) to flag another node to the whole mesh as stolen, when the stolen node receives this flag, it could do some stuff to "improve" the situation:

- sound off an alarm with a buzzer
- send position updates more often
- limit button features and user interaction
- display on screen: "<h2>Device Locked</h2>\nThis node has been flagged as stolen.", allow customization of the message

The device could also lock itself if it detects a position outside of a pre-determined range, but this could generate false positives, so it would be better for the user to monitor the device position and lock it with a button.

This feature should be disabled by default, and maybe implemented as a plugin. I think you would need some kind of authorization to flag a node as stolen, otherwise every mesh node could lock any other mesh node or be hijacked.

RicInNewMexico, 2022-01-30:
I would actually go a step further. Physical access is the ultimate weakness in any communications system.

I would actually suggest the ability to flag a device as a fixed-location node. 

Then, have a fixed node alert over the mesh on any unexpected movement (there would have to be some level of fuzziness to account for variability during startup / restart / gps loss / etc) AND have it automatically dump (if a confirmed movement over X time period) all configuration data / encryption. Maybe have a "dump channel on theft" flag that could be associated with individual channels so the user could choose which channels to dump. It should dump everything but the absolute minimum needed to continue position reporting. 

I think much of this would probably have to be a "down the road" goal since we have some significant bugginess with GPS location data on some device platforms. If implemented in the near term it would have to be a function that's disabled on devices with known buggy position data.

LolloDev5123, 2022-01-31:
The ability to remotely remove all critical data in a stolen node would be even better, it could be also another feature, and it would permanently delete all channels, etc. essentially making it so that if they control the node firmware-side they can't do anything nor see anything related to the mesh

mc-hamster, 2022-01-31:
There have been several similar proposals including this great one. Maybe the solution isn't to build a single solution but instead expand the functionality of the admin channel then allow someone on the network send commands to set the configuration they want.

closed 2022-02-04 by garthvh