meshtastic/firmware #1169: Feature Request: Add support the RAK14000 WisBlock E-Ink Display (RAK5005 base)

markbirss, 2022-01-31:
[image]

Product Features
Module Specification
3.3 V Input Voltage, On/Off Control by the WisBlock Core
Ultra-wide viewing angle
Ultra-low power consumption
Includes a Three-Button module
E-Ink Display and Three-Button module are connected over FPC (Flexible PCB)
White-Black
Display Resolution: 212 x 104 pixels
Size
Display Size: 2.9 inches
#

[image]

Product Link
https://store.rakwireless.com/products/wisblock-epd-module-rak14000?variant=39534109655238

Pinout
https://docs.rakwireless.com/Product-Categories/WisBlock/RAK14000/Datasheet/#overview-2

RAK14000 WisBlock E-Ink Display Overview
https://docs.rakwireless.com/Product-Categories/WisBlock/RAK14000/Overview/#product-description

RAK 5005 Base
https://docs.rakwireless.com/Product-Categories/WisBlock/RAK5005-O/Overview/#product-description

feh123, 2022-01-31:
This would be great to have. With the small size of the board I have wrecked a 5005-0 trying to solder an OLED screen!

markbirss, 2022-02-02:
@feh123 

I was able to get my 1.54 inch e-paper working, the resolution and screen size appears to be incorrect for the RAK14000 display...

It should be (tested with RAK code sample)
https://github.com/RAKWireless/WisBlock/tree/master/examples/common/IO/RAK14000-Epaper-Monochrome

DEPG DEPG_HP = {250,122,40,20,40,30,40,50,90,40}; //use DEPG0213BNS800F41HP as default

2.13 inch 250 x 122
SSD1680 driver

Also this display is not yet supported by GxEPD2 only a b/w/r version
https://github.com/ZinggJM/GxEPD2/commit/31e52bdeb3297af5b3db07b7412564055541bf69

markbirss, 2022-02-03:
Ok, so I have added support to what was the e-paper library for a similar e-paper with slightly different resolution but works.
250x128 vs 250x122
The interface layout needs adjustment

[image]

markbirss, 2022-02-03:
Anyone else want to test and help to fix the layout ?

Here is the repository 
https://github.com/markbirss/Meshtastic-device

uses
https://github.com/markbirss/GxEPD2

markbirss, 2022-02-21:
@garthvh 

This PR has been merged, i will send another PR to fix the T-ECHO platformio.ini to point to

[https](https://github.com/meshtastic/GxEPD2.git)

and not

https://github.com/geeksville/GxEPD2.git

https://github.com/meshtastic/Meshtastic-device/pull/1199

closed 2022-02-21 by markbirss