meshtastic/firmware #1170: "...time ago" stuck at "0 sec"

mchisari, 2022-01-31:
I've noticed this when displaying other node info screen: sometimes, under "Signal %" it starts ticking 0...1...2 seconds ago; some others, it stays at "0 seconds ago", without increasing. Why does it happen? 
(device is Heltec Lora V2.0 - no GPS)

mchisari, 2022-02-06:
I paired it with an android phone, but I see this behavior on Heltec's own display, client disconnected.

garthvh, 2022-02-08:
Do you have a device with GPS or wifi on the mesh?

mchisari, 2022-02-13:
I've elaborated a bit.
It seems this happens when the device was just rebooted, and it shows past seen nodes, before receiving any new message from them (or maybe received only a single location/announce message?).
I have two Heltec devices, so without GPS; one is paired with android phone with "provide location to the mesh" set; the other one is fixed and acts as a repeater, provided with a fixed location and never getting time locally, so if restarted it has to wait for "companion" to provide current time (at least I guess that's how it works).
This to say that, depending on what's happened before, M, F, or both could have unset time. 
It seems that one of these combinations ends up in M displaying "0 secs ago" constantly; other times, I've seen a more intriguing "456565 hours ago" (evidently, time from 1/1/1970). Maybe "0 secs ago" comes up when remote timestamp is past current time?
If so, maybe there should be an "invalid time" flag and/or special timestamp, so that these situation could be handled more meaningfully; for instance, upon restarting all known stations timestamp could be shown as "??? ago" until current time is set by GPS, application, or other nodes time sent.

garthvh, 2022-08-25:
Not seeing this behavior on 1.3

closed 2022-08-25 by garthvh