meshtastic/firmware #1179: Store & Forward Plugin Stacktrace after Crash

caveman99, 2022-02-02:
3 Nodes on the Network, one has the Store&Forward Plugin running. The Dump is from S&F node. Don't look for the commit hash, this is a private build.

```
PS C:\dev\embedded\mt\Meshtastic-device\bin> python .\exception_decoder.py -p ESP32 -e .\firmware-tlora-v2-1-1.6-1.2.53.01d534d-d.elf .\exception.txt
stack:
0x400ebea4: StoreForwardPlugin::historyAdd(_MeshPacket const&) at /builds/seven/meshtastic/Meshtastic-device/src/plugins/esp32/StoreForwardPlugin.cpp:184
0x400ec2e6: StoreForwardPlugin::handleReceived(_MeshPacket const&) at /builds/seven/meshtastic/Meshtastic-device/src/plugins/esp32/StoreForwardPlugin.cpp:277
0x400de665: MeshPlugin::callPlugins(_MeshPacket const&, RxSource) at /builds/seven/meshtastic/Meshtastic-device/src/mesh/MeshPlugin.cpp:128
0x400e1c81: Router::handleReceived(_MeshPacket*, RxSource) at /builds/seven/meshtastic/Meshtastic-device/src/mesh/Router.cpp:180
0x400e1d51: Router::perhapsHandleReceived(_MeshPacket*) at /builds/seven/meshtastic/Meshtastic-device/src/mesh/Router.cpp:180
0x400e1d71: Router::runOnce() at /builds/seven/meshtastic/Meshtastic-device/src/mesh/Router.cpp:180
0x400ddb75: ReliableRouter::runOnce() at /builds/seven/meshtastic/Meshtastic-device/src/mesh/DSRRouter.cpp:240
0x400d5492: concurrency::OSThread::run() at /builds/seven/meshtastic/Meshtastic-device/src/concurrency/OSThread.cpp:45
0x400f2531: ThreadController::runOrDelay() at /builds/seven/meshtastic/Meshtastic-device/.pio/libdeps/tlora-v2-1-1.6/Thread/ThreadController.cpp:153
0x400dd031: loop() at /builds/seven/meshtastic/Meshtastic-device/src/main.cpp:791
0x4011ab45: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
```

caveman99, 2022-02-02:
raw dump looks like this in the logfile:

```
[...]
23:17:17 952 [Router] Plugin 'text' considered
23:17:17 952 [Router] Plugin 'canned' wantsPacket=1
23:17:17 952 [Router] Plugin 'canned' considered
23:17:17 952 [Router] Plugin 'ExternalNotificationPlugin' wantsPacket=1
23:17:17 952 [Router] Plugin 'ExternalNotificationPlugin' considered
23:17:17 952 [Router] Plugin 'StoreForwardPlugin' wantsPacket=1 
23:17:17 952 [Router] --- S&F Received something
23:17:17 952 [Router] Packet came from - PortNum_TEXT_MESSAGE_APP
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ebea4  PS      : 0x00060a30  A0      : 0x800ec2e9  A1      : 0x3ffd34f0
A2      : 0x3ffd5db4  A3      : 0x3ffea630  A4      : 0x00000000  A5      : 0x3ffbb688
A6      : 0x00000000  A7      : 0x3ffd347c  A8      : 0x00000000  A9      : 0x3ffd34d0
A10     : 0x000e887f  A11     : 0x401eb028  A12     : 0x3ffd34f0  A13     : 0x3ffd34c0
A14     : 0x00000008  A15     : 0x3ffd345c  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x400ebea4:0x3ffd34f0 0x400ec2e6:0x3ffd3510 0x400de665:0x3ffd3530 0x400e1c81:0x3ffd3560 0x400e1d51:0x3ffd3580 0x400e1d71:0x3ffd35a0 0x400ddb75:0x3ffd35d0 0x400d5492:0x3ffd35f0 0x400f2531:0x3ffd3610 0x400dd031:0x3ffd3640 0x4011ab45:0x3ffd3660

Rebooting... 
```

mc-hamster, 2022-02-02:
> 3 Nodes on the Network, one has the Store&Forward Plugin running. The Dump is from S&F node. Don't look for the commit hash, this is a private build.
>
> ```
> PS [...]

Thank you for the report though the store and forward plug-in is not yet production ready. 

I’m pretty sure there is an off by one index error somewhere, at this point I’m gonna focus more on the next generation firmware before getting back to this.

closed 2022-02-02 by mc-hamster