meshtastic/firmware #1181: Feature Request: send http data back to end node

anudeepsharma254, 2022-02-03:
I want to use lora device(not a point to point it's a mesh) as a gateway, the end node is esp8266 and whenever it requires data, will send command through UART to LORA mesh(one Lora connects serially) and Lora which's having internet should get/post the data to server and send the replay(server data) back to, through mesh to esp8266.

mc-hamster, 2022-02-03:
May be possible to reuse the new functionality built into the S&F plugin for this, namely the message history to create a very minimal on device client that only works with high-spec esp32 devices.

sachaw, 2022-04-26:
This is a combination of https://github.com/meshtastic/Meshtastic-device/issues/1204 and some custom software, closing.

closed 2022-04-26 by sachaw