meshtastic/firmware #1204: Feature Request: MeshAPI over UART

caveman99, 2022-02-14:
Expose the client API over a (compile time configurable) serial interface. Either HW Serial Number or RX/TX Pins. This way another Microcontroller can access the API and facilitate the creation of a standalone device.

The Combination RAK5005 Base/RAK4631 Core/RAK2305 IO would be a first platform to implement this on, nRF52 core for meshtastic and an ESP32 for the application, doing screen and an I2C keyboard. 

Apart from that, this would be useful for any integration with existing applications, microcontroller or more high-level hardware.

geeksville, 2022-02-21:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/rak5860-for-mqtt-internet-connectivity/3464/8

mc-hamster, 2022-07-18:
> Expose the client API over a (compile time configurable) serial interface. Either HW Serial Number or RX/TX Pins. This way another Microcontroller can access the API and facilitate the creation of a [...]

Plumbing for this is in the serial plugin. Next step is to create a new instance of the phone api and pass it over. Very likely just need to mirror what's used now for the primary interface, minus the debug logging.

garthvh, 2022-08-05:
Should use the new protos mode https://github.com/meshtastic/Meshtastic-protobufs/blob/cb2fb77bd8c2751e82b1017cd7545789a037fb7d/module_config.proto#L88

garthvh, 2023-02-11:
Done, being used

PR #2060 (2022-12-22): ProtobufAPI in SerialModule

will resolve #1204
closed 2023-02-11 by garthvh