meshtastic/firmware #1215: Feature Request: incoming packets' RSSI through Python API/CLI

ub3aiw, 2022-02-15:
It would be extremely helpful to have incoming packets' RSSI available through Python API and/or CLI -- for antenna tuning and experimentation. For instance, as per SX1276/77/78/79 datasheet, both SNR and RSSI are required to accurately evaluate a received packet's signal strength (P. 87 [here](https://semtech.my.salesforce.com/sfc/p/E0000000JelG/a/2R0000001Rbr/6EfVZUorrpoKFfvaF_Fkpgp5kzjiNyiAbqcpqh9qSjE?__hstc=212684107.1346a557561f33f4b505dadcadb95b84.1644953215825.1644953215825.1644953215825.1&__hssc=212684107.3.1644953215825&__hsfp=3719723820)). Moreover, I might be mistaken, but I'm not seeing any references to SNR being used alone, uncoupled from RSSI.

garthvh, 2022-02-20:
Pretty sure this is not available for all supported radios.

mc-hamster, 2022-02-20:
Closing this issue. Please look at SNR instead of RSSI. They are functionally equivalent except SNR takes into account the noise floor

closed 2022-02-20 by mc-hamster