meshtastic/firmware #1218: [Bug]: tbeam hangs and crashes after a period of time

Vertikar, 2022-02-17:
### Category

Other

### Hardware

T-Beam

### Firmware Version

1.2.55.9db7c62

### Description

Device is just left alone and eventually hangs after a few hours.

--info output
Owner: Karl TBeam 4dd8 (KT4)
My info: { "myNodeNum": 4145958360, "hasGps": true, "numBands": 20, "firmwareVersion": "1.2.55.9db7c62", "rebootCount": 62, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 4145958360, 'user': {'id': '!f71e4dd8', 'longName': 'Karl TBeam 4dd8', 'shortName': 'KT4', 'macaddr': '30:c6:f7:1e:4d:d8', 'hwModel': 'TBEAM'}, 'position': {'batteryLevel': 100, 'time': 1645107096}, 'lastHeard': 1645107096}  {'num': 3328166159, 'user': {'id': '!c65fc90f', 'longName': 'Karl T-Echo', 'shortName': 'KT', 'macaddr': 'c3:4b:c6:5f:c9:0f', 'hwModel': 'T_ECHO'}, 'position': {'latitudeI': -, 'longitudeI': , 'time': 944010489, 'latitude': -, 'longitude': }, 'lastHeard': 1643936916, 'snr': 6.25}  {'num': 4064765348, 'user': {'id': '!f24765a4', 'longName': 'Karl TBeam 65a4', 'shortName': 'KT6', 'macaddr': '08:3a:f2:47:65:a4', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': -, 'longitudeI': , 'time': 1643936716, 'latitude': -, 'longitude': }, 'lastHeard': 1643936746, 'snr': 5.75}

Preferences: { "positionBroadcastSecs": 1200, "phoneTimeoutSecs": 900, "lsSecs": 43200, "region": "ANZ", "positionBroadcastSmart": true, "isRouter": true }

### Relevant log output

```Shell
3:45:11 3638 [RadioIf] Completed sending (id=0x73ccc53a Fr0xa4 To0xff, WantAck0, HopLim2 Ch0xb1 encrypted rxtime=1645105506 rxSNR=3.75 priority=64)
13:45:16 3643 [PositionPlugin] Sending pos@0:6 to mesh (wantReplies=0)
13:45:16 3643 [PositionPlugin] cancelSending id=0x67cf47dd, removed=0
13:45:16 3643 [PositionPlugin] Providing time to mesh 1645105516
13:45:16 3643 [PositionPlugin] Update DB node 0xf71e4dd8, rx_time=1645105516
13:45:16 3643 [PositionPlugin] handleReceived(LOCAL) (id=0x67cf47df Fr0xd8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 rxtime=1645105516 priority=10)
13:45:16 3643 [PositionPlugin] No plugins interested in portnum=3, src=LOCAL
13:45:16 3643 [PositionPlugin] Add packet record (id=0x67cf47df Fr0xd8 To0xff, WantAck0, HopLim3 Ch0x0 Portnum=3 rxtime=1645105516 priority=10)
13:45:16 3643 [PositionPlugin] Should encrypt MQTT?: 1
13:45:16 3643 [PositionPlugin] Expanding short PSK #1
13:45:16 3643 [PositionPlugin] Installing AES128 key!
13:45:16 3643 [PositionPlugin] enqueuing for send (id=0x67cf47df Fr0xd8 To0xff, WantAck0, HopLim3 Ch0xb1 encrypted rxtime=1645105516 priority=10)
13:45:16 3643 [PositionPlugin] txGood=22,rxGood=20,rxBad=0
13:45:16 3643 [PowerFSM] lsEnter begin, ls_secs=43200
13:45:16 3643 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:16 3643 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:16 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:16 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3644 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3645 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:17 3645 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:18 3645 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:18 3645 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:18 3645 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:18 3645 [PowerFSM] radio wait to sleep, txEmpty=0
13:45:18 3645 [RadioIf] Starting low level send (id=0x67cf47df Fr0xd8 To0xff, WantAck0, HopLim3 Ch0xb1 encrypted rxtime=1645105516 priority=10)
13:45:18 3645 [RadioIf] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=25, time 2760 ms
13:45:18 3645 [RadioIf] AirTime - Packet transmitted : 2760ms
13:45:18 3645 [PowerFSM] GPS prepare sleep!
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4012765c: e0a89681 ff5f0100 ffffffff
Core 1 register dump:
PC      : 0x40127662  PS      : 0x00060733  A0      : 0x6a700000  A1      : 0x3ffd3630
A2      : 0x00000000  A3      : 0xd94dcc24  A4      : 0x3ffbf730  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3ffc9efc  A8      : 0x40082f3c  A9      : 0x3ffd3610
A10     : 0x0c4cdf1c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ff000e4
A14     : 0x3ffc9d28  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x40127662:0x3ffd3630 0x6a6ffffd:0x3ffd3660 0x400d3fab:0x3ffd3680 0x400efe82:0x3ffd36a0 0x400dd019:0x3ffd36c0 0x400d5272:0x3ffd36e0 0x400f65d1:0x3ffd3700 0x400dd384:0x3ffd3730 0x40121871:0x3ffd3750

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4009662c: 0b126608 21033370 2920aab0
Core 1 register dump:
PC      : 0x40096630  PS      : 0x00060033  A0      : 0x800966a7  A1      : 0x3ffd34f0
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000009  A5      : 0x3ffd34c8
A6      : 0x3ffc1138  A7      : 0x3ffc1141  A8      : 0x3ff4808c  A9      : 0x8001f880
A10     : 0xc001f880  A11     : 0x00000000  A12     : 0x00000009  A13     : 0x00000030
A14     : 0x3ff5f068  A15     : 0x3ff48070  SAR     : 0x0000001b  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x40096630:0x3ffd34f0 0x400966a4:0x3ffd3510 0x400901bd:0x3ffd3530 0x400904da:0x3ffd3550 0x4008217a:0x3ffd3570 0x4012765f:0x3ffd3630 0x6a6ffffd:0x3ffd3660 0x400d3fab:0x3ffd3680 0x400efe82:0x3ffd36a0 0x400dd019:0x3ffd36c0 0x400d5272:0x3f
fd36e0 0x400f65d1:0x3ffd3700 0x400dd384:0x3ffd3730 0x40121871:0x3ffd3750

continues here: https://pastebin.com/0aCaXNF0
```

garthvh, 2022-02-17:
How long have you been running meshtastic on this device?

Vertikar, 2022-02-18:
Hi Garth, have had it for a while now, but only recently noticed the crashes.
Looks like it crashed overnight, but I ran --info after I rebooted it so looks like didn't get all the same logs from last time, just got this on the serial console

```
[tio 01:20:49] Connected
"Z;
eG"+/'5}X=bE@PhYZW
"       !f24765a4Karl TBeam 65a4KT6:e05>s=bE@Ph8Z6
"2FV5~X=XbEPh8Z6
"2FV5X=FV5X=mYbE@Ph8Z6
"V5X=bEMb5j^=bEPh8Z6
"FV5X= ZbE@Ph8Z6
"V5X=bE@Ph8Z6
GV5X=E@PhPre BT: 215508 heap size
                                 [D][esp32-hal-i2c.c:1345] i2cProcQueue():  Gross Timeout Dead start=0xe3d27, end=0xe3d27, =0, max=50 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbe3dc
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffbbfc8
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffbc24c bits=10
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffbac9c
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffd95c0
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
Shutdown BT: 215508 heap size

[tio 12:33:45] Disconnected
```

mkinney, 2022-03-02:
This *might* be fixed with https://github.com/meshtastic/Meshtastic-device/pull/1266

Once there is a new release, please test and let us know if the issue is resolved or not.

mkinney, 2022-03-02:
Please try https://github.com/meshtastic/Meshtastic-device/releases/tag/v1.2.57.f7c6955

Vertikar, 2022-03-07:
Can't seem to replicate it while plugged in anymore, but it does seem to occasionally hang while disconnected on battery.
Of course nothing visible in the serial log then

Vertikar, 2022-03-21:
Was still seeing a similar hang on 1.2.57, have just upgraded to 1.2.59 so shall see how that goes.

Vertikar, 2022-03-24:
had my tbeam up for 44 hours on 1.2.59, but it seems to have crashed again.
Will see if I can reproduce and get any logs

Vertikar, 2022-03-24:
Crashed pretty quickly this time, output doesn't seem very useful though:

```
01:55:29 2440 [RadioIf] Starting low level send (id=0x706f4c6d Fr0xd8 To0xff, WantAck0, HopLim3 Ch0xb1 encrypted rxtime=1648086927 priority=10)
01:55:29 2440 [RadioIf] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=63, time 4726 ms
01:55:29 2440 [RadioIf] AirTime - Packet transmitted : 4726ms
01:55:29 2440 [PowerFSM] GPS prepare sleep!
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
93Cp609fx0 :   9x4012767c: 0361a202 ff5f0100 ffCr00x0    A   01003 : S 000f 0 a5dfx:932GurIE
                 000f30C
604x 013f30:a004x 0a3f3G  swd1a1   1fGM tsm2b     A  9fcff8 :  41SE0 :: E0cx:d48040 04dGd u p60e    AA   1cf00 :  5AE0x0::F0k3x:d290x  a3f3x:8Gu etm:2CP  7a0003::  01AA 260RVEEL 0
                                     00ff33x:   AA  0:  T
020:AGNC                                                 0B041004x 013f30:7604x 063f30:Guoaio 2rCP   000f0 :  11A
 c1 D DO   A0004xi0c2f30:2604x  42f3x:b48040 03dfx:a69040 04dfx:Gurlor02e PA  000fx0 :  21
020:AGNC2dfx:d48040 0edf30a26040 02df30:b004x 043fGM1tsm2b     A  9fcff8::  41SE0 :: E0cx:d48040 0adf30a66040 0edf30:b000x  03f3x:7600x  63f3x:Guoaio 9rCP   000f0 :   1A
 36040 09df304b004x 002f3x:7690x  62f3x:419040 23dfx:a19GurIx
                                                             63Cp609dx0 :
 c1 D DOe4x0     A  c81000:: S00000e0:62f30:a004x  22f3x:d290x  e2f3x:d69040 0edf30d48040 01dGd u p60r    AA  81cf00 :  5AE0x ::F0k3x:d200x  e2f3x:2690x  42f3x:b48040 0cdfx:a69040 0fdf30Gurlor02e PA  000fx0 :  21
3690x  42f3x:b4804  02dfx:a69040 6adfx0419040 0ddf30a76GuoIx
e:   2AA  11400 ::  RX0x0 :F0tfx:a19040 01df30d69040 04df30d42040 07df30a26040 0adf30Gurlny02e SAA 000d30 :   3dfx:419040 08df30a19040 0bdf30:a004x
069040 0fdf30d42040 022f30:2604x 052f30:b004x 042f30:76GuoIcM67C:036100x     A   3d00AEL  0:0040 0ddf30a2604x 002f30:b004x 042f30:7604x  a2f3x:a000x
000:DGNC   A00040l0c1f3x:Guoaio 9rCPA  000f0 ::  2Af30a69040 0Gtioa  Ctx0 :   5 85dfxGreI  00GdpuccrCu0003 ::   8A                                  Gne)l41C 008x    AA   f0000 :E 0c00S0 0bdf304b004x 0e1f30:7604x  a2f3x:a090x  62f3x:d190x  22Gipcu 0
        0
 002f33x    AA  x:  Tu swd1a1   1
                     0B0419040 04df30a0Gu:leM2 C
                                                 3a1000x:   1AA  190DCBL   0904  63dfx:d69040 05dfGM tau2a
                                                                                                                 97ff38:   1AS 0::
000d3x0     A  00R  N                                                                                                               0a0:b000x  41f3x:7600x  a1f3xGreIn 00   0A
 :a004x  e1f3x:d200x  a1f3x:2690x
                                 Gne.l41C 008x:   6AA  80000 : 0c00S0 0e0f30:b000x  01f3x:7690x  61f3x:4190
Gicnn02Ct4x0      7AA 8000x0  02c0H00 80ffx:d48040 0ddf30a2604x 001f30:b004x 031f30:7604x                  GE  .03Cu660f0::   8A
020:D DO           c8000x:  U00000e0e9df30d69040 0bdf30:d204x 0a0f3x:2690x  01f3x:b480x  c1dGe1rsm6br    AA  81cf0 ::  5AE00 : L0c30:d204x 090f30:2600x  00f3x:b48040 cfdfx:a69040 03df30Guoaor 9rCPA  000f0 ::  21
        0
         3604x 000f30:b004x  c0f3x:7690x0 20dfx:419040 01df30a16GuoIx
                                                                     67C:306d0x0:
036000x0    A   0d00AEL   :0040 00df30a26040 04df30:b004x 070f30:7604x 060f3x:a090x3xGreIny00   0A 000d3x0    3A  30D  U3x:a19040 ecdfx:a19040 0fdf30:a004x
000f33x0    A  70R  NC00x0  0rf30:7690x  60ffx:419040 23dfx:a19040 06df30d69040 09df3Gr n  b0   0A  6AA  f0000 :E 0c00S0 0fcf30:b004x 000f30:7604x  60f3x:a000x  20f3x:d290x  e0Gipcua0
 :a090x  2fcfx:a19040 01df30d60040 04df30d42040 070f30:2Gu:leM2 C
                                                                  3ab000x:   1AA  f90DCBLL  09040 0ccf30:a004x 06ff3x:d290x  20f3x:2690x  80ffx:d480
0e20A0x 0aff3x:d290x  2dcfx:d69040 00df30dd204x 030f30:2604x 0Gtion02Ct4x :                                                                         GE ..03Cu6003 ::   8A
93Cp609fx0 :   990x  afcfx:d69040 0ecf30:d204x  e0f3x:2Gu epec CP  720003:   1AA  660DCBLL 006040 06cf30:a004x  2ff3x:d190x  eccf30d6004x 0fff30:d20GurIEdGd u p60e:   2A   1cf00 :  5AE0x0 :F0k3x:d19040 6acfx:d69040 0cff30:d290x  a0ffx:d69040 03dfxGreIny00   0A
                 000f40C
Gne)l41Cr00x0    AA   f000 ::E 0c00S0 02cf30410040 05cf30:7604x  eff3x:a19040 abcf30a16040 02Gtatht8
31818x0    C004x0f0rcfx:a69040 04cf30419040 06cf30:7604x  2ff3x:419040 0dcfGM tau2a                 i ::   4A
020:D DO                                                                                  9fff38:   1AS 0 :  E0ax:b080x  cfcfx:a69040 05cf3041004x 08ff30:7604x  6ff3x:Guoaor 9rCPA  000f0 ::  21
        0
0e20A00 0ccf30:d204x  6ff3x:269040 c2cfx:d48040 05cf30a2604x 0Gtion0 Ctx0 :    b000
000ef3x0    AA  x:  N                                                           188000x:  U00000l0ebcfx:d69040 0def30:d290x  aff3x:d69040 04cf30d48040 06cGd u p60e:   2AA  1cf00 :   RE0x0 :F0kfx:a19040 0ccf30:a004x  2ef3x:d190x  e2cfx:d69040 05cf3Gu n  b0    1
690x  aecfx:419040 0bcf30:7604x  eef3x:419040 a1cf30a16GuoIxtm:2oPP  00003 :   1AA 260RVEEL
818800x0   A0004xi0cef30:2600x  4ef3x:d48040 09cf30a6604x 0cef3x:b080x  4fcGe rsp6br    AA  8cff0 ::  5SE00 : L0c30dd204x 02ef3x:2600x  8effx:d48040 0acf30a2604x  cef3xGreIny02  SAA 000c3x0    3
[... cut ...]
63Cp609fx0 :    40 03cf30a16040 06cf30:a004x  6bf3x:d19GurIEtm:2CP  700003 :  01AA e60RVEEL
Gne)l41Cr0080    AA   f0000 :E 0c00S0 0ccf30:b004x  8af3x:7690x0 e3cfx:419040 05cf30a76040 02Gtatht8
 c1 D DO   C004x0f0acfx:a69040 0ecf30:b000x  cbf3x:a69040 04cfx0419040 07cfGM tau2a     A  9fff38::  1AS 0 :  E0ax:b48040 0bcf30a2604x  4af3x:b080x  0bcfx:a69040 05cf30Gurlor09rCPA  000fx0 :  21
Gicnn02Ce4x0     A  c80000:: U00000e0e5cf30:a004x  eaf3x:d290x  abcfx:d69040 0ecf30:d209890?0Guru Me0f 0@3 1x:2690x0 8acfx:d48040 0eaf30:2604x  cbf3x:b480
```

garthvh, 2022-03-24:
Are you connecting the device to a phone or actually using it as a router?

Vertikar, 2022-03-24:
Just a router, want to have it permanently mounted in the car eventually.
It crashed again about an hour ago, but recovered itself. 

```
12:43:32 2453 [PowerFSM] radio wait to sleep, txEmpty=0
12:43:32 2453 [RadioIf] Starting low level send (id=0x65e66a27 Fr0xd8 To0xff, WantAck0, HopLim3 Ch0xb1 encrypted rxtime=1648125810 priority=10)
12:43:32 2453 [RadioIf] (bw=125, sf=12, cr=4/8) packet symLen=32 ms, payloadSize=63, time 4726 ms
12:43:32 2453 [RadioIf] AirTime - Packet transmitted : 4726ms
12:43:32 2453 [PowerFSM] GPS prepare sleep!
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
000d3x0     A  b00000l0e4dfx:0Gu gtm:2CP   00003 :  01AA e600x0 : 0tf3:a19040 05Gtatna8
31818x0    CE  60504x 023f30:7604x 0aGaniht8                                           i ::   4A
020004x0f0adfx:a69040 0fdf304b004x 02df30a7604x 0aGtaiht8cff8::  41SE 0c00S0 0e2f30:b000x  03f3x:7690x  63f3x:Guoair 9rCPA  000f0 ::  21
840f8x0    CE  60504x  02f3x:7690x  62ffx:419040 02dfx:a19040 05Gtatna8
                                                                       i :   3AA 31818x0    C
i :   3AA 31818x0    C0c2f3x:d690x0 82dfx:d48040 0ddf30a2604x 003f30:b004x 083fGM1tsm2b     A  97cff8 :  41SE 0c00S0 0c2f3x:b000x  82f3x:7690x  e2dfx:419040 0fdf30a16040 03Gtatna00x0 :F0tfx:a19040 0bdf30d69040 0edf30:d204x 023f30:2604x  83f3xGreIny02   0A
GE .d03Cu9003 :   7AA 80003x0 D DO  090x  a5dfx:d69040 08dfx:d48040 0adf30a2604x 0e2f30:b000
3600x  42f3x:b080x  02ffx:a690x0 62dfx:419040 0ddf30a76GuoIx
                                                            67C:306d0x0     A  93f000:  T
                                                                                         0B0410040 03df30a76040 062f30:a004x 062f30:d290x  22f3x:dGu epec2CP  760003::  01AA 2600x0 :F0kfx:a19040 22dfx:d69040 05df30d48040 08df30a69040 0bdf30Gurlny02e SAA 000dx0 :   3
                                                                                                                                                                                                                                                                          c10004xi0c1f30:2604x  c2f3x:b080x  82f3x:7690x  e2f
3x:419040 a2dfG  sau1a
818800x0   AEL  060040 0bdf30a66040 0fdf30:b004x 002f30:7604x  62f3x:a000x:   5 0edf30419040 00df30a76040 04df30:a004x 0e2f30:d204x 0a2Gd cu 6
93Gu n61ffx:419040 0edf30a16040 01df30:a004x 04df30:d20Gur(E              Gne)l41C 008x:   AA   f0000 :EVEEL
g :   3AA 3181x0 :   C 06Giptna0x  01ffx:d48040 c0dfx:a69040 0Ganoa  Cs0 ::   5
                       0f02040 a1dfx:a0Gu:leM20C
GE .d03Cd9003 :   7AA 80003x :D DO               3a3000x    AA   19000:: L0c30d48040 04df301GuCgio:2oPP  0000f ::  11A
3604x  01f3x:b000x  c1f3x:bGu epe:2CP  760003::  01A  1600x0 :F0k3x:d19040 21dfx:d69040 84df3:d38GuoIx
                                                                                                      67C:306d0x0:
                                                                                                                        93f000:  T
0e20A0x 0b0f30:d200x  a0f3x:d690x0 01ffx:d48040 c4dfx:1Gu gtm:2CP   00003 :  01AA 260000 : 0tf30a16040 0cdf30:a004x  a0f3x:d290x  61f3x:2690x0 c1ffG  tau2a     a004x  e1fGe1rsm6b     A   8cff8 :  41SE 02c0H0  00f3x:b080x  c0dfx:a69040 03dfx0419040 06dGd u  60e:   2AA  1c400 ::  RX
 0f020x  20f3x:d190x  ebdfx:d69040 0edf30d42040 011f30:2604x 0Gtion0 Ct40 :                                                                                      008ff3x:   1AS  0000 0 29dfx:419040 0cdf30a76040 060f30:a000x  21f3x:d290x  e1Gipcu 0
0 d.e00Cd0083:   6AA  8000x0 :AGNC                                              188000x:  UXLL  :004x 020f30:2604x  80f3x:b080x  40dfx:a69040 00df3041904
 69040 06df30410040 09df30:7604x  e0f3x:a090x  a0dfx:a19GurIx
                                                             63Cp609fx0 :
0                                                                             000f00x:  N
 :419040 a0dfx:a19040 69dfx:d69040 0bdf30dd204x 0e0f30:2Gu epec C
0000f4x0R  N   940 00df30a16040 04df30da004x 06df30:d20Gur(Ex 0e0f30:a004x:   AGd u p60e:   AA   1cf00 :  5AE2600x  00f3x:b080x  c0dfx:a69040 0edf30Gurlor09rCPA  000fx0 :  21
 :a000x  efffx:a19040 00df30d60040 03df30dd2040 050f30:2Gu:lpec C
                                                                  3aa000x:   1AA  b6000 ::L0k30:d290x  efcfx:d69040 0fcf30:d204x 020f30:2604x  80f3xGueIny02e SAA 000c3x0    3
                                                                                                                                                                                20004xl0cff3x:2690x0 8fcfx:d48040 0eff30:2604x  c0f3x:b080x  80fGe1rsm6br    AA  8cff0 ::  5SE 02c0H0  cfffx:d48G d.e00Cd9003::   7AA 8000x0
:DGNC0 0acf30:b000x  0fcfx:a69040 00df3041004
                                  0
G d.d03Cd9003::   7AA 8000x0 :D DOx  0ff3x:b080x  cfcfx:a69040 0acf304b004r    AA  8ccf0 ::  5AE 02c0H00 8fffx:d48040 04cf30a2604x 07ff30:b000x  8ff3x:a69040 0Gtioa  Ctx0 :   5
                                  0
0 d.e00Cd0083:   6AA  8000x0 :AGNC0 0def30:b000x  8ff3x:a69040 04cf3041904     A  9fff38::  11S  0c00S0 0bcf30:b000x  0ecfx:a69040 01cf3041004x 04ff30:7604x  aGaniht Cs0 :   4A
G d.e00Cd9003::   7AA 8000x0 :DGNC0 07cf304b004x 04ef3x:7690x0 adcf3041904     A  9bcff8::  41SE 0c00S0 00ef3x:b000x  ceffx:a69040 0bcf3041004x 00ef3x:7690x  6Gania  Cs0 ::   5
                                  0
[... cut ...] fffffffffffffffffffffaa0a?00B????????????????????0003abd7d4da
ad4daad4daad4da3333333333333333333333300000000990089999899998999986H??:??:?? 0

//\ E S H T /\ S T / C

??:??:?? 0 booted, wake cause 0 (boot count 1), reset_reason=taskWatchdog
??:??:?? 0 Filesystem files:
??:??:?? 0   /static/index.69df4442.js.gz
??:??:?? 0   /static/favicon-16x16.png.gz
??:??:?? 0   /static/Logo_Black.svg.gz
??:??:?? 0   /static/android-512.png.gz
??:??:?? 0   /prefs/db.proto
??:??:?? 0   /static/robots.txt.gz

```

garthvh, 2022-06-15:
Closing as no further work is being done on 1.2