meshtastic/firmware #1222: [Bug]: sending factory_reset to RAK5005 does not reset the device to factory resets

mkinney, 2022-02-19:
### Category

Serial

### Hardware

Rak4631

### Firmware Version

1.2.55.9db7c62

### Description

1. Connect RAK 4631 5005 to usb port
2. Run this command:

```
meshtastic --set factory_reset true
```

3. Expected the device to reset. 

```
(venv) sweet Meshtastic-python % meshtastic --set factory_reset true
Connected to radio
Set factory_reset to true
Writing modified preferences to device
(venv) sweet Meshtastic-python % meshtastic --info
Connected to radio

Owner: Unknown f46f (?6F)
My info: { "myNodeNum": 863106159, "hasGps": true, "numBands": 13, "firmwareVersion": "1.2.55.9db7c62", "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8 }
Nodes in mesh:  {'num': 863106159, 'user': {'id': '!3371f46f', 'longName': 'Unknown f46f', 'shortName': '?6F', 'macaddr': 'f1:9c:33:71:f4:6f', 'hwModel': 'RAK4631'}, 'position': {}}  {'num': 4067915087, 'user': {'id': '!f277754f', 'longName': 'Unknown 754f', 'shortName': '?4F', 'macaddr': 'dc:c6:f2:77:75:4f', 'hwModel': 'RAK4631'}, 'position': {'latitudeI': 457317000, 'longitudeI': -1226697628, 'altitude': 75, 'batteryLevel': 65, 'time': 1645305522, 'latitude': 45.7317, 'longitude': -122.6697628}, 'lastHeard': 1645305757, 'snr': 4.75}  {'num': 2475227164, 'user': {'id': '!9388f81c', 'longName': 'Unknown f81c', 'shortName': '?1C', 'macaddr': '44:17:93:88:f8:1c', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': 457316864, 'longitudeI': -1226699377, 'time': 1645305466, 'latitude': 45.7316864, 'longitude': -122.66993769999999}, 'lastHeard': None, 'snr': 6.25, 'lastReceived': {'from': 2475227164, 'to': 4294967295, 'id': 953329315, 'rxSnr': 6.25, 'hopLimit': 3, 'rxRssi': -35, 'fromId': '!9388f81c', 'toId': '^all'}, 'hopLimit': 3}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "US", "positionFlags": 35, "cannedMessagePluginAllowInputSource": "a", "cannedMessagePluginMessages": "aa", "cannedMessagePluginSendBell": true }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ

```

Notes:

1. meshtastic --version
1.2.84

2. mac

garthvh, 2022-03-04:
Looks like the application flash area could be wiped using the code here https://github.com/beegee-tokyo/WisBlock-API/blob/main/src/flash-nrf52.cpp 

I verified that installing the AT api does wipe the meshtastic application flash and reset the device

mkinney, 2022-03-17:
I think this fixed in 1.2.58

closed 2022-03-17 by mkinney

DooMMasteR, 2022-11-29:
So I flashed the firmware-t-echo-2.0.6.97fd5cf.uf2 to my T-ECHO because meshtastic was not working after Upgrading and it is still not working, some weird settings and such.  
But I could set a custom name and flashing the reset file does nothing to that data.  
Yes the T-ECHO does not boot anymore, but after flashing the current firmware back to it, the settings (custom name) are back again an my T-Echo is inoperable.  
It works via serial, but Bluetooth is broken.  
I have a second T-ECHO which I upgraded straight away and that one is not experiencing issues, but I flashed the file to verify, and yes, all settings just survive.