meshtastic/firmware #1228: Add ability to change speed/paylod type via CLI for serial plugin

D4rk4, 2022-02-20:
Please add the ability to change speed via meshtastic CLI.
It's needed for example to connect the Brother EP-44 typewriter as a messenger device to a meshtastic network.
Old serial printers like this EP typewriter can operate at 110,300,1200 baud rate (common is 300 bps).
So we need an easy way to change the baud rate without rebuilding firmware.
Also will be fine to have the ability to reconfigure port payload type (set PortNum_TEXT_MESSAGE_AP).

mc-hamster, 2022-02-20:
Good idea.

Note: The speed should be selectable as an enum.

RicInNewMexico, 2022-03-01:
I went ahead and assigned this to myself to work on since I'm already eyes deep in the serial plugin. This should be achievable by release of the 1.3.x breaking release.

D4rk4, 2022-04-12:
Any progress?

D4rk4, 2022-04-19:
@mc-hamster related to your pull (#1329), it's not possible to set the speed to 300 or 1200, the minimum supported speed by firmware, for now, is 2400...

garthvh, 2022-09-11:
baud rates for 1.3

	case baudDefault = 0
	case baud110 = 1
	case baud300 = 2
	case baud600 = 3
	case baud1200 = 4
	case baud2400 = 5
	case baud4800 = 6
	case baud9600 = 7
	case baud19200 = 8
	case baud38400 = 9
	case baud57600 = 10
	case baud115200 = 11
	case baud230400 = 12
	case baud460800 = 13
	case baud576000 = 14
	case baud921600 = 15

caveman99, 2022-09-12:
While we're at it, can we hook the second part of the request (Also will be fine to have the ability to reconfigure port payload type (set PortNum_TEXT_MESSAGE_APP).) to the serial plugin mode settings? Right now we only have simple, default (the same) and proto (not wired yet). That would be low hanging fruit and a good first issue, though it requires a proto change / addition.

garthvh, 2022-09-12:
> While we're at it, can we hook the second part of the request (Also will be fine to have the ability to reconfigure port payload type (set PortNum_TEXT_MESSAGE_APP).) to the serial plugin mode [...]

We should get the protos in, as we need to freeze them for real with the release of 1.3.41

closed 2022-10-16 by garthvh