meshtastic/firmware #1229: [Bug]: ProtobufPlugin: Unhandled exception on malformed protobuf messages

oe1rfc, 2022-02-20:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

1.2.55

### Description

If a malformed protobuf message could not be parsed in  `handleReceived()` an error is logged, but the plugin's `handleReceivedProtobuf` will be called with `decoded=NULL`:
https://github.com/meshtastic/Meshtastic-device/blob/master/src/mesh/ProtobufPlugin.h#L67-L73

Since a lot of plugins don't check for a decoded payload, this leads to a panic probably by nullpointer dereference:
```
??:??:?? 56 [Router] Error: can't decode protobuf reason='string overflow', pb_msgdesc 0x0x3f4050bc
??:??:?? 56 [Router] Error decoding protobuf plugin!
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

According to the docs, `decoded` may be `NULL` for `handleReceivedProtobuf()` (eg. when handling different portnums), I'll open a PR to check for a nullpointer in the plugins `handleReceivedProtobuf`, or should we rather abort if decoding fails in `handleReceived()`?

thebentern, 2022-04-26:
Related to #1407

thebentern, 2022-10-17:
Pretty sure we've fixed this with some of the recent safeguards in 1.3.x

closed 2022-10-17 by thebentern