meshtastic/firmware #1262: [Bug]: Wifi password hardcoded as 'sekrit'

SigmaScott, 2022-03-01:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

v1.2.56.596a73c 

### Description

When using the meshtastic flasher to config a device the password is set to 'sekrit' even after setting the password to correct password.

RicInNewMexico, 2022-03-01:
Is it failing to connect to your wifi network? (stepping through the screens on your OLED display using the middle button should show a screen with wifi connection stats or alternatively you should see it listed as a client if you log into your wifi router and look through the DHCP leases). 

None of the clients have the ability to retrieve a stored password from the device so they all display the password as 'sekrit' if a password is set. Is this what you are seeing or does it seem like it's actually setting it to 'sekrit' and causing it not to connect?

SigmaScott, 2022-03-01:
yes it is failing to connect.  I noticed this when going back in to turn on the mqqt it quit working.  if I change it via the cli tools to the correct password it connects again.

SigmaScott, 2022-03-01:
I have reset device to factory and been able to re create this.
1.factory reset, reset device
2. start meshtastic flasher
3. click on device setting
4. on first page set short name and long name click save
5. on wifi enter ssid and password click save
6. on cli where flasher was started you can see loging that password and ssid like I set them
7. restart device, it will not connect
8. go back into flaher and see the seekr it in the password field.  not sure it this is what was acutlly set then read from device or if this is what was set bgut it is filled again because of logic in program

mc-hamster, 2022-03-01:
@SigmaScott thanks for the report. 

@RicInNewMexico sekrit is being set in: https://github.com/meshtastic/Meshtastic-device/blob/2a02b4594f5fa7067dfa5d6c55f5618f98974801/src/modules/AdminModule.cpp#L190

That's being called by: https://github.com/meshtastic/Meshtastic-device/blob/2a02b4594f5fa7067dfa5d6c55f5618f98974801/src/modules/AdminModule.cpp#L78

@mkinney I think this is a bug in either flasher or the python script. Some reason, it's calling into "AdminMessage_set_radio_tag" which is resetting the password.