meshtastic/firmware #1264: [Bug]:  meshtastic --test failing, two fresh heltec-v2.0 c1.2.52 units loaded

bradshawlupton, 2022-03-01:
### Category

Other

### Hardware

Heltec v2

### Firmware Version

1.2.52

### Description

I bare metal build, erasing everything, loading the stock firmware, then Platformio building the git tagged version.

I did not check the stock firmware, I will do that next.  I built the git tagged version with no alterations.

The units, part of a 4 node ham mesh network all talk beautifully, 

setting up MQTT today. 

sensors DHT11 and DS18B20 work perfectly

tried the tests, then looked in test.py, did not see any simple debugging to add, so reported here.

tnx.

### Relevant log output

```Shell
sh-3.2# meshtastic --test
INFO file:test.py openDebugLog line:142 Writing serial debugging to log_dev_cu.usbserial-0001
INFO file:test.py openDebugLog line:142 Writing serial debugging to log_dev_cu.usbserial-3
Connection changed: meshtastic.connection.established
INFO file:test.py testAll line:163 Ports opened, starting test
INFO file:test.py testThread line:125 Found devices, starting tests...
INFO file:test.py runTests line:95 Running 5 tests with wantAck=True
Connection changed: meshtastic.connection.established
ERROR file:test.py runTests line:108 Test 1 failed, expected packet not received (1 failures so far)
ERROR file:test.py runTests line:108 Test 2 failed, expected packet not received (2 failures so far)
ERROR file:test.py runTests line:108 Test 3 failed, expected packet not received (3 failures so far)
ERROR file:test.py runTests line:108 Test 4 failed, expected packet not received (4 failures so far)
ERROR file:test.py runTests line:108 Test 5 failed, expected packet not received (5 failures so far)
ERROR file:test.py runTests line:118 Too many failures! Test failed!
Connection changed: meshtastic.connection.lost
Connection changed: meshtastic.connection.lost
Warning: Test was not successful.
sh-3.2#
```

thebentern, 2022-03-01:
Heltec devices have been notorious flakey on the tests, so I'm not sure there is a clear _bug_ here to solve.

bradshawlupton, 2022-03-01:
absolutely fine!  I am doing formal software builds for repeatability.  

Any suggestion on getting more debugging into test.py?  To make it more developer friendly 

Thanks, Bradshaw,

mkinney, 2022-03-01:
Well, I must admit that if I get *one* failure, I don't consider that a failure. You got 5 failures, which is not good.

What OS? Are both devices Heltec? If you're building from -device, what is the git commit that you're using? What -python version are you using?

I just tested the latest -python on Ubuntu.

Here's what I just tested:

```
$ meshtastic --test
INFO file:test.py openDebugLog line:142 Writing serial debugging to log_dev_ttyUSB0
INFO file:test.py openDebugLog line:142 Writing serial debugging to log_dev_ttyUSB1
Connection changed: meshtastic.connection.established
INFO file:test.py testAll line:163 Ports opened, starting test
INFO file:test.py testThread line:125 Found devices, starting tests...
INFO file:test.py runTests line:95 Running 5 tests with wantAck=True
Connection changed: meshtastic.connection.established
INFO file:test.py runTests line:112 Test 1 succeeded 1 successes 0 failures so far
INFO file:test.py runTests line:112 Test 2 succeeded 2 successes 0 failures so far
INFO file:test.py runTests line:112 Test 3 succeeded 3 successes 0 failures so far
INFO file:test.py runTests line:112 Test 4 succeeded 4 successes 0 failures so far
ERROR file:test.py runTests line:108 Test 5 failed, expected packet not received (1 failures so far)
ERROR file:test.py runTests line:118 Too many failures! Test failed!
Connection changed: meshtastic.connection.lost
Connection changed: meshtastic.connection.lost
Warning: Test was not successful.
```

meshtastic --version
1.2.87

meshtastic --info --port /dev/ttyUSB0 shows: "firmwareVersion": "1.2.53.19c1f9f" and 'hwModel': 'HELTEC_V2_1'
meshtastic --info --port /dev/ttyUSB1 shows: "firmwareVersion": "1.2.56.596a73c" and 'hwModel': 'HELTEC_V2_1'

garthvh, 2022-09-11:
Can't replicate on 1.3, 1.2 is EOL

closed 2022-09-11 by garthvh