meshtastic/firmware #1281: [Bug]: TBeam Stuck In Boot Loop

yowierocket, 2022-03-07:
### Category

Serial

### Hardware

T-Beam

### Firmware Version

1.2.57.f7c6955

### Description

Tbeam was working the previous day; started it this morning and it is now in a boot loop.

### Relevant log output

```Shell
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
ets Jun  8 2016 00:22:57
```

mc-hamster, 2022-03-08:
Hi!

thanks for the log. That indicates a hardware issue with the SPI flash. You may try reinstalling meshtastic onto the device, if it doesn’t work, I don’t think there’s much we can do. 

I will leave this issue open until we hear back from you.

yowierocket, 2022-03-08:
Thanks for the response,

Gave re-flashing it a go. Gave an error (expected) when detecting the device but was still able to re-flash it. Device is now fully functional, thank you!

closed 2022-03-08 by yowierocket

mc-hamster, 2022-03-08:
Woohoo! Thanks for writing back!