meshtastic/firmware #1300: Enhancement: Send projected, labelled marker by tapping in map

Kollensperger, 2022-03-16:
would it be possible to implement sending a location on the map, as a named marker displayed on the map, without having a device at that specific location? It would be great to be able to e.g define a common meeting point that appears on everyone’s map without having to be at that point already.

This is surely a ton of work but would be a useful addition at some point in the future

mc-hamster, 2022-03-22:
Sounds like @Kollensperger 's request is to be able to send out location waypoints. That can likely be a variant of the message payload. No need to go lower level than that and its persistence can be treated like a received message.

garthvh, 2022-03-22:
> Sounds like @Kollensperger 's request is to be able to send out location waypoints. That can likely be a variant of the message payload. No need to go lower level than that and its persistence can be treated like a received message.

So could we just have an optional position on a message payload? This would allow me to only send one packet for the location pin feature.

mc-hamster, 2022-03-22:
Oh, that's a possibility and would be even more elegant than what I was thinking.

Let me take a look at the protos later today. This should be easy.

garthvh, 2022-08-24:
This is fully implemented on device now, can be sent from messages on iOS and can be received in the webui

closed 2022-08-24 by garthvh

MaximKrasikov, 2022-09-08:
will i be able to send list of markers to other users via waypoint ? Do I need to fix the firmware of the device and proto for this?

garthvh, 2022-09-12:
They are individual packets like messages, there is no centralized list at this time.