meshtastic/firmware #1307: Enhancement: HTTP Backend Authorization

LolloDev5123, 2022-03-19:
When accessing an endpoint like the webpanel, authenticate the IP for N hours before asking again,
[image]

The webserver that meshtastic uses supports this, here's an example https://github.com/fhessel/esp32_https_server/blob/master/examples/Authentication/Authentication.ino

garthvh, 2022-03-20:
Appears to use http basic authentication, is this correct?

sachaw, 2022-03-20:
Not a fan of basic auth, would prefer a session cookie

mc-hamster, 2022-03-20:
Same. Really would prefer not using this. 

Kevin has proposed handling sessions within protobufs. Let’s explore that.

sachaw, 2022-03-20:
I like that, some kind of "Auth" module:
Takes a key/password in one message,
returns a session token
Then the client can store it how they like and refresh at x interval

sachaw, 2022-03-20:
Maybe make it a part of the initial config step, the default is blank so it will work without providing one.

closed 2022-03-22 by garthvh