meshtastic/firmware #1314: GPS change proposals

sachaw, 2022-03-22:
We currently use two libraries for parsing GPS messages:
- Sparkfun GPS library (Deprecated)
- TinyGPS++
Whilst there is a Sparkfun library that supersedes that of which we use, only generation 8 and onwards are supported

Whilst considering the upgrade, waying up the benefits of being able to parse UBX messages but only on certain models doesn't make a whole lot of sense.

Although UBX is extremely efficient and allows much higher update rates, the nature of the Meshtastic project likely does not benefit from this.

Further testing is needed, but we may find that only parsing NMEA messages is an elegant solution.

mkinney, 2022-03-22:
I'd like to see a prototype or POC of using TinyGPS++, if we can do that.

mkinney, 2022-03-22:
What are the requirements? In other words, what Meshtastic supported devices have GPS?

The Meshtastic devices with GPS are:
* RAK 4631/5005 with 12500  in slot X
* RAK 4631/5005 with 1910 in slot X
* T-Beam
* T-Echo

mkinney, 2022-03-22:
https://github.com/meshtastic/TinyGPSPlus - forked repo
http://arduiniana.org/libraries/tinygpsplus/ - nice "getting started" code/explanations

caveman99, 2022-03-22:
my only question mark is if TinyGPS++ will work with the I2C Ublox device on the 19003. We might need to keep the Sparkfun lib around for that.

caveman99, 2022-03-31:
I have made some tweaks to my locak TinyGPS Fork and it's looking slightly better. Now if my T-Echo would aquire a fix i could check for jumping coordinates and narrow that down ...

https://github.com/caveman99/TinyGPSPlus/tree/patch-1

caveman99, 2022-04-06:
Just a note, the RAK1910 is EOL.

thebentern, 2022-04-26:
@sachaw @caveman99 is this essentially done? Can we close this issue?

closed 2022-04-26 by sachaw