meshtastic/firmware #1341: WiFi AP isn't working in 1.2.59

mc-hamster, 2022-03-31:
WiFi AP isn't working in 1.2.59

Both admin mode and regular AP isn't functioning right.

Need to diagnose to determine root cause.

Also need to see if this is happening in 1.3.x

igorparsadanov, 2022-04-15:
appears to be not working in 1.2.53.19c1f9f either.

sachaw, 2022-05-12:
My observations from 1.3.x:

 - New firmware flashed
 - SSID & PSK set via CLI
 - device auto-reboots
 - 50/50 if the device decides to generate a tls cert or not
 - subsequent reboots or re-issuing the cli command fixes it (appears that settings stick just won't always gen cert)
 - WiFi is now working
 - Further reboots are 50/50 if the network will reconnect or just keep spamming `STA:Disconnected`
 - Reflashing appears to be the only reliable workaround.

garthvh, 2022-06-22:
Seems to be working in 1.3

garthvh, 2022-08-04:
Working in 1.3, 1.2 is closed to updates

closed 2022-08-04 by garthvh