meshtastic/firmware #1345: [Bug]: Issues setting psk key via serial

rwcts, 2022-03-31:
### Category

Serial

### Hardware

T-Beam, Heltec v2.1

### Firmware Version

1.2.52

### Description

This includes versions 1.2.59 and 1.2.58.

When programming a new Heltec or TBeam unit via serial using the below commands it fails. (Note COM9 has to be used as there are other internal devices that use com ports. This is done via Phython install on Windows 11 64bit.

meshtastic --ch-set psk random --ch-index 0 --port COM9
meshtastic --ch-set psk KEYHERE --info --ch-index 0 --port COM9

The only way to get the key in is to attempt to use the client.meshtastic.org site using serial and then paste the key into the gui. Using the client.meshtastic.org site sometimes takes several times to connect it but it eventually works and saves the changes. 

### Relevant log output

```Shell
C:\Users\User>
C:\Users\User>meshtastic --ch-set psk random --ch-index 0 --port COM9
Connected to radio
Writing modified channels to device

C:\Users\User>meshtastic --ch-set psk KEYHERE --info --ch-index 0 --port COM9
Connected to radio
Aborting due to: 'KEYHERE' has type str, but expected one of: bytes
```

mkinney, 2022-03-31:
You cannot just set the `KEYHERE` to a string value. It must be in a specific notation "bytes" and of a specific length, depending on which level you want.

From the `--help`, here is an example:

```
To set encryption with your own key: '--ch-set psk 0x1a1a1a1a2b2b2b2b1a1a1a1a2b2b2b2b1a1a1a1a2b2b2b2b1a1a1a1a2b2b2b2b --ch-index 0'
```

You probably just want to use 'random' instead of 'KEYHERE'.

rwcts, 2022-03-31:
> You cannot just set the `KEYHERE` to a string value. It must be in a specific notation "bytes" and of a specific length, depending on which level you want.
>
> From the `--help`, here is an example: [...]

Yes good point. KEYHERE was just a placeholder. The key that we are using was actually generated by the meshtastic device when we were playing with the units when they first arrived. There is an actual string used.