meshtastic/firmware #1352: [Bug]:  Heltec v.2.1 Firmware 1.3.4.2b20fb3 OLED Displays wrong modem config

rwcts, 2022-04-04:
### Category

Other

### Hardware

Heltec v2.1

### Firmware Version

1.3.4.2b20bf3

### Description

The Heltec Lora 32 v.2.1 devices was loaded with the alpha version 1.3.4.2b20bf3 and on the second (OLED Screen) screen it shows the channel as modem config as MedSlow when looking at the output of the --info shows it to be LongFast.

The OLED display is not showing the proper modem config/speed setting.

(History of unit) prior to the firmware being loaded on this unit it had a valid configuration and was 1.2.59 and in the advanced options of the Meshtastic Flasher v1.3alpha. 13 was update only. After the flash this is what was noticed. We then reflashed the unit WITHOUT update only selected and it still showed the second screen with MedSlow

### Relevant log output

```Shell
Connected to radio

Owner: Unknown 5588 (?88)
My info: { "myNodeNum": 2225100168, "firmwareVersion": "1.3.4.2b20bf3", "rebootCount": 2383, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 2225100168, 'user': {'id': '!84a05588', 'longName': 'Unknown 5588', 'shortName': '?88', 'macaddr': '78:21:84:a0:55:88', 'hwModel': 'HELTEC_V2_1'}, 'position': {}}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "positionFlags": 35 }

Channels:
  PRIMARY psk=default { "modemConfig": "LongFast", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAiIBAQ

---------after the flash without using update only selected ------------

Connected to radio

Owner: Unknown 5588 (?88)
My info: { "myNodeNum": 2225100168, "firmwareVersion": "1.3.4.2b20bf3", "rebootCount": 2, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 2225100168, 'user': {'id': '!84a05588', 'longName': 'Unknown 5588', 'shortName': '?88', 'macaddr': '78:21:84:a0:55:88', 'hwModel': 'HELTEC_V2_1'}, 'position': {}}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "positionFlags": 35 }

Channels:
  PRIMARY psk=default { "modemConfig": "LongFast", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAiIBAQ
```

garthvh, 2022-04-04:
Settings are lost between 1.2 and 1.3, MedFast is the new default

rwcts, 2022-04-04:
> Settings are lost between 1.2 and 1.3, MedFast is the new default

When running meshtastic --ch-set modem_config LongSlow --ch-index 0 --port COM9
 it shows ShrtFast on the screen when --info shows LongSlow but still reporting incorrectly on the second screen as ShrtFast

rwcts, 2022-04-04:
I also wanted to add that when looking at client.meshatastic.org via serial it also shows MedFast in the GUI  when the --info shows LongSlow

andrekir, 2022-04-04:
please paste the result of `meshtastic --support`

rwcts, 2022-04-04:
> please paste the result of `meshtastic --support`

Below are the system details from --support (Windows 11 64bit)

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10-10.0.22000-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.3a11
 Executable: C:\Users\User\AppData\Local\Programs\Python\Python310\Scripts\meshtastic
 Python: 3.10.3 CPython MSC v.1929 64 bit (AMD64)

Please add the output from the command: meshtastic --info

caveman99, 2022-04-04:
This was fixed with https://github.com/meshtastic/Meshtastic-device/pull/1326 and should be correct in 1.3.5 - if not please reopen

closed 2022-04-04 by caveman99