meshtastic/firmware #1363: I2C screen detection is inconsistent with certain OLED models

campbellcole, 2022-04-07:
I recently received 3 T-BEAMs with the upgraded LoRa and GPS chips. After flashing the devices, all three screens produced random, but consistent, garbage pixels. I have fixed this issue by modifying the `i2cScan` file, however my changes would break the detection mechanism.

The OLED screens I purchased are made for the T-BEAM, but they're not the old 0.96" model. They're a new SH1106 model that LilyGO sells as well. [Here is the link](https://www.aliexpress.com/item/1005003510957337.html).

Initially, no matter what I did, the screens would display garbage. Then, a few days ago, somebody committed a change that added a new subtype to the detection mechanism for SH1106. This subtype only appears from my screens ONLY when the device has been powered on, but was just reset. So currently, the firmware as it is REQUIRES a reset after every power on to get the screen working.

The problem is that this device reports two different subtypes depending on when you ask for it. When initially booting after first receiving power, these OLEDs report a subtype of `0x06`, which is supposedly reported from SSD1306 as well. Then, once reset using RST, the subtype becomes `0x08` consistently. 

I fixed this by moving the `0x06` condition into the SH1106 detection condition, but since it was previously in the SSD1306 detection, I don't think submitting a pull request is a good idea. Instead, I wanted to let you know of this discrepancy because I understand it may not be worth fixing. 

Thanks, let me know if you need any more information about this.

garthvh, 2022-04-07:
This work is in progress in 1.3

caveman99, 2022-04-08:
Since i wrote that code i'll revisit that section soon. I have the exact same model of lilygo shield you linked and it reliably gives me 0x08 on the scan, no matter previous state.

campbellcole, 2022-04-08:
> Since i wrote that code i'll revisit that section soon. I have the exact same model of lilygo shield you linked and it reliably gives me 0x08 on the scan, no matter previous state. 

That's fascinating. I initially used a stable build and it didn't work at all, then I saw that commit you made and wondered if it was because of that OLED. I figured it was because it partially fixed the problem.

I don't know hardly anything about i2c though and I have no idea how this result is even possible. My only guess is that it was reading the wrong data from i2c because the stream started after the cpu was expecting, but I don't know if that's how the protocol works. 

Regardless, thanks for this amazing software. I hope to contribute more once I familiarize myself with the codebase.

caveman99, 2022-04-22:
@campbellcole can you try again with a recent master build?

campbellcole, 2022-04-30:
somehow it is able to get 0x06 twice in a row on first boot and still goes with ssd1306, resetting still fixes this

mc-hamster, 2022-04-30:
I’ve seen reports of this on and off over the last two years. What would be an elegant solution?

campbellcole, 2022-04-30:
> I’ve seen reports of this on and off over the last two years. What would be an elegant solution?

I don't know what level of accessibility this project is shooting for, but personally I'd prefer a simple `#define` for SH1106 vs SSD1306. I can implement this if you feel it is the right approach.

I believe this is how it worked before the i2c scan impl, it's odd that the response is so inconsistent but until that is better understood I can't imagine any fixes for the current solution. Again, not an expert with i2c though.

caveman99, 2022-05-01:
These defines exist already alongside the autodetect mode to support device configurations with a known and defined set of hardware like the nano-g1 board

caveman99, 2022-05-01:
Look in src/graphics/screen.h to find them.

garthvh, 2022-08-04:
I2C detection is working great in 1.3 now

closed 2022-08-04 by garthvh