meshtastic/firmware #1364: Pinetab/Pinephone

joshbowyer, 2022-04-08:
The other Pinetab issue is locked so Im making this to add this here; somebody has written and released the driver for these devices:

https://codeberg.org/JF002/pinedio-lora-driver

garthvh, 2022-09-12:
https://github.com/meshtastic/Meshtastic-device/issues/614#issue-776806484

> while working on pinetab support I noticed that the radio lib spi operations are super inefficent. Both for regular register operations and bulk packet read/writes.
> 
> The fix is to use the more modern two buf implementations of transfer() provided on the nrf52/esp32 (rather than single byte transfers). This will let the hardware SPI controllers do more of the work. This would also prevent the linux port from doing lots of extra USB operations/kernel entries.

Djhg2000, 2023-07-14:
Doing some preliminary work on this, so far I've come up with a few notes:

* This seems to be the recommended driver for the CH341A USB-SPI bridge: https://github.com/rogerjames99/spi-ch341-usb . It works but could conflict with any other CH341 (as in the ones without the "A" at the end) based devices.

* There's another driver that looks promising and it's supposedly written to be upstreamable, but it has a blocking bug for using it with the PineDio USB adapter: https://github.com/frank-zago/ch341-i2c-spi-gpio/issues/5 .

* The LoRa chip on the USB adapter is an SX1262, not a plain RF95.

* This demo code works, tested between the USB adapter and my PinePhone Pro with the LoRa back (needed some minor patching to work; it expects the pogo pins to be on I2C bus 2 but on PPP they're on I2C bus 5). Demo code: https://codeberg.org/JF002/pinedio-lora-driver

* I'm looking into making a RadioLob HAL for PinePhone (Pro), looks to be easy enough with the Raspberry Pi HAL as a starting point. No promises on when I'll have something to work with.

thebentern, 2023-07-14:
I think this is the way to go, long term.
>     * I'm looking into making a RadioLob HAL for PinePhone (Pro), looks to be easy enough with the Raspberry Pi HAL as a starting point. No promises on when I'll have something to work with.

I'd be interested to see what you come up with. It's been quite some time since I've fought the Pinedio USB adapter. I've got a couple here. The last thing I remember battling was linux kernel version specific drivers and libgpio. Reach out if you have any questions or run into anything you think we can help with. I'd love to see this working, it was just taking too much of my bandwidth in my last attempt.

Djhg2000, 2023-07-17:
The major hurdle nowadays seems to be that out of tree driver for the SPI part. There's the one that works with the demo code and that's what I'm working with for now, but I'd like to move to the more upstreamable driver when it becomes ready.

I have some previous experience with the Linux SPI and GPIO subsystem APIs but I'm not as familiar with the I2C subsystem API yet. Pretty much no experience with VSCodium though so it took me a while to set it up for Meshtastic, but at least it can compile the T-Beam firmware now.

My main issue is just finding the time to work on it so I don't want to make any promises on deadlines.

thebentern, 2023-07-18:
> The major hurdle nowadays seems to be that out of tree driver for the SPI part. There's the one that works with the demo code and that's what I'm working with for now, but I'd like to move to the more upstreamable driver when it becomes ready.
> 
Sounds like a solid plan!
> 
> My main issue is just finding the time to work on it so I don't want to make any promises on deadlines.

Yeah, I get this. Ultimately it's why I bounced off of this one and focused more on other aspects of Meshtastic. It was a big time sink and there were higher priority things to tackle. 😅

Djhg2000, 2023-07-25:
Well, here's some unfortunate news; https://github.com/rogerjames99/spi-ch341-usb no longer compiles since Linux 6.4 with the following (first) error:
```
/home/djhg2000/install/spi-ch341-usb/spi-ch341-usb.c:212:30: error: field ‘gpio’ has incomplete type
  212 |     struct gpio_chip         gpio;                              // chip descriptor for GPIOs
      |
```

This means I'll have to boot into Linux 6.3 to continue development and it won't compile on vanilla Debian Sid without reverting to an older kernel anymore. So with this in mind I'll take a break from this a couple of weeks and enjoy my vacation, hopefully the new driver is ready enough when I get back home. If it isn't I'll keep developing with Linux 6.3 just to get the bulk of the HAL ready and update whatever needs updating later. The new CH341A driver should in theory be compatible as far as the APIs go but the GPIO mappings and SPI configuration sequences are different.

I don't feel comfortable releasing what little I have yet because (1) it's too incomplete to be of any use and (2) it's continuously undergoing restructuring as I figure out how to best match up the APIs of RadioLib and Linux. I think I have a fairly complete `init()` and `term()` implementations but I'll still have to figure out a good way of translating GPIO pin numbers to line objects in `pinMode()` and the likes.

This is further complicated by how I thought I could write the HAL in mostly pure C rather than C++ (which I am a lot less experienced with), so there's some extra legwork involved there too. The main complication there, aside from mixing C and C++ code, is that `libgpiod` is written in pure C with anonymous enums. C allows anonymous enums where they are roughly equivalent to `#define` statements but C++ limits the scope of anonymous enums in a fundamentally incompatible way (they don't reach into classes), so I need to use the C++ bindings and rewrite the whole thing in C++. Not a huge problem but it's yet another thing that slows progress.

Djhg2000, 2023-11-23:
I've made some progress on this using the new upstreamable driver (or rather, using this WIP branch of the extended GPIO configuration https://github.com/mr-nice/ch341-i2c-spi-gpio/tree/pinedio ) and I've pushed what I have at the moment to this WIP repo https://github.com/Djhg2000/RadioLib/tree/linux-generic-wip

It's still missing the code for handling interrupts, I have some ideas on how to do that with threads but I have yet to implement that functionality. The rest of the code should work but I haven't tested the HAL itself beyond compiling it because I haven't had the time to set up separate testing hardware without a radio.

Anyway, progress is still being made and I haven't abandoned by efforts. :)

jp-bennett, 2023-11-25:
This is great news, @Djhg2000 as we've just recently made some great strides in getting Meshtasticd running on Raspberry Pi hardware with a real radio. Hoping to have some good collaboration between these two efforts, and eventually a combined solution.

jp-bennett, 2023-12-30:
Taking a quick look at this. We may run into trouble with the Chip Select pin getting claimes by the SPI driver. In my Raspberry Pi work, I've found that radiolib needs access to that pin for things to work. I'm going to see about building the kernel module on a dev machine this weekend, and trying it out.

jp-bennett, 2024-01-01:
@Djhg2000 Hey, I may have something working. Would love for you to take a look. It's not 100% acting right, but I have something. This is using Meshtastic master, the native-unity branch, target native.

git clone https://github.com/frank-zago/ch341-i2c-spi-gpio.git
cd ch341-i2c-spi-gpio
make
sudo insmod ch341-core.ko
sudo insmod gpio-ch341.ko
sudo insmod i2c-ch341.ko  
sudo insmod spi-ch341.ko

echo "spidev 1" | sudo tee /sys/class/spi_master/spi0/new_device
echo spidev | sudo tee /sys/bus/spi/devices/spi0.1/driver_override
echo spi0.1 | sudo tee /sys/bus/spi/drivers/spidev/bind
sudo ln /dev/spidev0.1 /dev/spidev0.0

In /etc/meshtasticd/config.yaml
Lora:
 Module: sx1262
 CS: 0
 IRQ: 10
 Busy: 11

Djhg2000, 2024-01-16:
@jp-bennett Sorry for not getting back on this sooner. I can't find the branch "native-unity" in this repo and attempting to build the "native" target from the master branch results in the following errors:

```
Building in release mode
Using meshtastic platformio-custom.py, firmware version 2.2.19.fd8b1687
Compiling .pio/build/native/libca0/LovyanGFX/lgfx/v1/platforms/sdl/Panel_sdl.cpp.o
Compiling .pio/build/native/src/SerialConsole.cpp.o
Compiling .pio/build/native/src/airtime.cpp.o
Compiling .pio/build/native/src/buzz/buzz.cpp.o
Compiling .pio/build/native/src/concurrency/BinarySemaphoreFreeRTOS.cpp.o
Compiling .pio/build/native/src/concurrency/BinarySemaphorePosix.cpp.o
Compiling .pio/build/native/src/concurrency/InterruptableDelay.cpp.o
Compiling .pio/build/native/src/concurrency/Lock.cpp.o
Compiling .pio/build/native/src/concurrency/LockGuard.cpp.o
Compiling .pio/build/native/src/concurrency/NotifiedWorkerThread.cpp.o
Compiling .pio/build/native/src/concurrency/OSThread.cpp.o
Compiling .pio/build/native/src/detect/ScanI2C.cpp.o
Compiling .pio/build/native/src/detect/ScanI2CTwoWire.cpp.o
Compiling .pio/build/native/src/gps/GPS.cpp.o
Compiling .pio/build/native/src/gps/GeoCoord.cpp.o
Compiling .pio/build/native/src/gps/NMEAWPL.cpp.o
Compiling .pio/build/native/src/gps/RTC.cpp.o
Compiling .pio/build/native/src/graphics/EInkDisplay2.cpp.o
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../common.hpp:50,
                 from .pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/Panel_sdl.cpp:25:
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../arduino_default/common.hpp:39:31: error: ‘long unsigned int lgfx::v1::millis()’ was declared ‘extern’ and later ‘static’ [-fpermissive]
   39 |   static inline unsigned long millis(void)
      |                               ^~~~~~
In file included from .pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/Panel_sdl.hpp:23,
                 from .pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/Panel_sdl.cpp:21:
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:47:17: note: previous declaration of ‘long unsigned int lgfx::v1::millis()’
   47 |   unsigned long millis(void);
      |                 ^~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../arduino_default/common.hpp:44:31: error: ‘long unsigned int lgfx::v1::micros()’ was declared ‘extern’ and later ‘static’ [-fpermissive]
   44 |   static inline unsigned long micros(void)
      |                               ^~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:49:17: note: previous declaration of ‘long unsigned int lgfx::v1::micros()’
   49 |   unsigned long micros(void);
      |                 ^~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../arduino_default/common.hpp:49:22: error: ‘void lgfx::v1::delay(long unsigned int)’ was declared ‘extern’ and later ‘static’ [-fpermissive]
   49 |   static inline void delay(unsigned long milliseconds)
      |                      ^~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:51:8: note: previous declaration of ‘void lgfx::v1::delay(long unsigned int)’
   51 |   void delay(unsigned long milliseconds);
      |        ^~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../arduino_default/common.hpp:54:15: error: ‘void lgfx::v1::delayMicroseconds(unsigned int)’ was declared ‘extern’ and later ‘static’ [-fpermissive]
   54 |   static void delayMicroseconds(unsigned int us)
      |               ^~~~~~~~~~~~~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:53:8: note: previous declaration of ‘void lgfx::v1::delayMicroseconds(unsigned int)’
   53 |   void delayMicroseconds(unsigned int us);
      |        ^~~~~~~~~~~~~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../arduino_default/common.hpp:59:23: error: redefinition of ‘void* lgfx::v1::heap_alloc(size_t)’
   59 |   static inline void* heap_alloc(      size_t length) { return malloc(length); }
      |                       ^~~~~~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:55:23: note: ‘void* lgfx::v1::heap_alloc(size_t)’ previously defined here
   55 |   static inline void* heap_alloc(      size_t length) { return malloc(length); }
      |                       ^~~~~~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../arduino_default/common.hpp:60:23: error: redefinition of ‘void* lgfx::v1::heap_alloc_psram(size_t)’
   60 |   static inline void* heap_alloc_psram(size_t length) { return malloc(length); }
      |                       ^~~~~~~~~~~~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:56:23: note: ‘void* lgfx::v1::heap_alloc_psram(size_t)’ previously defined here
   56 |   static inline void* heap_alloc_psram(size_t length) { return malloc(length); }
      |                       ^~~~~~~~~~~~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/../arduino_default/common.hpp:61:23: error: redefinition of ‘void* lgfx::v1::heap_alloc_dma(size_t)’
   61 |   static inline void* heap_alloc_dma(  size_t length) { return malloc(length); } // aligned_alloc(16, length);
      |                       ^~~~~~~~~~~~~~
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:57:23: note: ‘void* lgfx::v1::heap_alloc_dma(size_t)’ previously defined here
[... cut ...]
.pio/libdeps/native/LovyanGFX/src/lgfx/v1/platforms/sdl/common.hpp:76:10: note: previous definition of ‘struct lgfx::v1::FileWrapper’
   76 |   struct FileWrapper : public DataWrapper
      |          ^~~~~~~~~~~
*** [.pio/build/native/libca0/LovyanGFX/lgfx/v1/platforms/sdl/Panel_sdl.cpp.o] Error 1
```

I can't see any other branch with the same purpose so I'm not sure where to go from here. Is is still relevant for me to keep working on the RadioLib side or are we moving forward with PlatformIO (which from what I can tell uses Arduino library abstraction layers) instead?

markbirss, 2024-01-16:
> @Djhg2000 Hey, I may have something working. Would love for you to take a look. It's not 100% acting right, but I have something. This is using Meshtastic master, the native-unity branch, target [...]

Modified for my DIY CH341 SX1262

cd ch341-i2c-spi-gpio
sudo insmod ch341-core.ko
sudo insmod gpio-ch341.ko
sudo insmod i2c-ch341.ko
sudo insmod spi-ch341.ko

echo "spidev 1" | sudo tee /sys/class/spi_master/spi11/new_device
echo spidev | sudo tee /sys/bus/spi/devices/spi11.1/driver_override
sudo mv /dev/spidev11.1 /dev/spidev0.0

[image]

`INFO  | 14:16:50 89 Rebooting
Portduino is starting, HWID=1, VFS root at /root/.portduino/default
Setting up Meshtastic on Portduino...
��@INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  /.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (53 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (168 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (94 Bytes)
DEBUG | ??:??:?? 0  /. (0 Bytes)
INFO  | ??:??:?? 0 Scanning for i2c devices...
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=37, swver=2.2.19.6284f4f
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 22
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded saved config version 22
/root/.portduino/default/prefs/module.proto does not exist
INFO  | ??:??:?? 0 No /prefs/module.proto preferences found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
/root/.portduino/default/oem/oem.proto does not exist
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
WARN  | ??:??:?? 0 Using nodenum 0xddb09507 
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 3, using EU_868
DEBUG | ??:??:?? 0 region=3, NODENUM=0xddb09507, dbsize=2
DEBUG | ??:??:?? 0 Read RTC time as 1705414611
DEBUG | ??:??:?? 0 Running without GPS.
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
/root/.portduino/default/prefs/cannedConf.proto does not exist
INFO  | ??:??:?? 0 No /prefs/cannedConf.proto preferences found
INFO  | ??:??:?? 0 CannedMessageModule is enabled
WARN  | ??:??:?? 0 SX126xInterface(cs=0, irq=6, rst=2, busy=4)
DEBUG | ??:??:?? 0 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO  | ??:??:?? 0 Starting meshradio init...
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 0 Radio freq=869.525, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 0 Set radio: region=EU_868, name=LongFast, config=0, ch=0, power=27
INFO  | ??:??:?? 0 Radio myRegion->freqStart -> myRegion->freqEnd: 869.400024 -> 869.650024 (0.250000 mhz)
INFO  | ??:??:?? 0 Radio myRegion->numChannels: 1 x 250.000kHz
INFO  | ??:??:?? 0 Radio channel_num: 1
INFO  | ??:??:?? 0 Radio frequency: 869.525024
INFO  | ??:??:?? 0 Slot time: 42 msec
[... 45 lines cut ...]
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Installing AES128 key!
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] enqueuing for send (id=0x6b8b456a fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=1)
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] txGood=1,rxGood=0,rxBad=0
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Using channel 0 (hash 0x8)
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Expanding short PSK #1
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Using AES128 key!
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] Installing AES128 key!
DEBUG | ??:??:?? 45 [DeviceTelemetryModule] decoded message (id=0x6b8b456a fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 priority=1)
DEBUG | ??:??:?? 45 [RadioIf] Starting low level send (id=0x6b8b456a fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=1)
DEBUG | ??:??:?? 45 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=37, time 509 ms
DEBUG | ??:??:?? 45 [RadioIf] AirTime - Packet transmitted : 509ms
GPIOPin::callISR(GPIO6, 6)
DEBUG | ??:??:?? 45 [RadioIf] Completed sending (id=0x6b8b456a fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted priority=1)
GPIOPin::callISR(GPIO6, 6)
DEBUG | ??:??:?? 72 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=25, time 419 ms
DEBUG | ??:??:?? 72 [RadioIf] Lora RX (id=0x3c81d681 fr=0xec to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=5.75 rxRSSI=-89)
DEBUG | ??:??:?? 72 [RadioIf] AirTime - Packet received : 419ms
DEBUG | ??:??:?? 72 [Router] Add packet record (id=0x3c81d681 fr=0xec to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxSNR=5.75 rxRSSI=-89)
DEBUG | ??:??:?? 72 [Router] Using channel 0 (hash 0x8)
DEBUG | ??:??:?? 72 [Router] Expanding short PSK #1
DEBUG | ??:??:?? 72 [Router] Using AES128 key!
DEBUG | ??:??:?? 72 [Router] Installing AES128 key!
DEBUG | ??:??:?? 72 [Router] decoded message (id=0x3c81d681 fr=0xec to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxSNR=5.75 rxRSSI=-89)
DEBUG | ??:??:?? 72 [Router] handleReceived(REMOTE) (id=0x3c81d681 fr=0xec to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=3 rxSNR=5.75 rxRSSI=-89)
DEBUG | ??:??:?? 72 [Router] Module 'position' wantsPacket=1`

Djhg2000, 2024-01-17:
> @Djhg2000 That branch got merged to master. You're hitting a bug in the LovyanGFX library that I did. Uninstall your system SDL development packages. That will change how the LovyanGFX library compiles, and avoid the error. I think the next release of Lovyan will have a fix.
> 
> The way Meshtastic supports running on Linux is the Portduino/PlatformIO libraries. That seems to work rather well, except my Pinedio has a problem with sending long packets when using the default long fast preset. I'm wondering if the lack of TCXO is causing the problem.

That issue seems to be an erroneous include path then, glad a fix in is the queue.

I managed to compile the daemon (output binary is `.pio/build/native/program` on my end but I guess that's expected?) and I can start it, but it seems I'm stuck in SimRadio mode. I've tried specifying `--config=/etc/meshtasticd/config.yaml` and I've tried adding `spidev: spidev0.0` after your config parameters as well, but it just seems to ignore it.

I tried using https://github.com/meshtastic/firmware/blob/master/bin/config-dist.yaml as a base and adding your config in the `Lora:` section as well but nothing seems to work. I've looked through the debug output in the console and I don't see it probing for any SPI interface. From the log posted by @markbirss I guess I should see some messages with a `SX1262` prefix?

Would it be possible to have some more detailed debug output regarding the radio init? Mostly interested in if it's trying to open a file descriptor to something in `/dev/` and which one. An option to exit with an error or non-zero return value if only SimRadio is available would speed up my testing a little but it's mostly a convenience thing until I write more scripts for setting up a test environment.

jp-bennett, 2024-01-17:
> Would it be possible to have some more detailed debug output regarding the radio init?

Yes, I can add some additional debug messages that should help you figure out what's going on.

jp-bennett, 2024-01-22:
@Djhg2000 Just added some debugging to master. You'll want to add ```Logging:
  DebugMode: true
``` to your config.yaml. The main things that were added is a message at basically the beginning of the log output, indicating what config file is getting used. And then when trying to init the radio, a debug message gets printed indicating what kind of radio, and which SPI port is being used. Hopefully that helps. Let me know how it goes.

thebentern, 2024-02-17:
This is essentially complete now with the burgeoning native linux effort and proven CH341 support.

closed 2024-02-17 by thebentern