meshtastic/firmware #1366: [Bug]: Using serial with client.meshtastic.org and OLED - its not displaying proper primary channel preset as --info

rwcts, 2022-04-08:
### Category

Serial

### Hardware

T-Beam

### Firmware Version

1.2.59.d81c1c0

### Description

The tbeam v1.1 M8 was programmed via python. Then after the programming was completed I visited client.meshtastic.org and used the chrome serial connection to look at the config. Under settings + Primary Channel, it shows that the preset was MidSlow which is not what the output shows when --info is made.
Channels:
  PRIMARY psk=secret { "modemConfig": "LongSlow",

Also, the screen on the device shows "ShrtFast" on the OLED

So which one is correct?

### Relevant log output

```Shell
--info (partial output shown)

Channels:
  PRIMARY psk=secret { "modemConfig": "LongSlow", "psk": "my real key hidden goes here", "uplinkEnabled": true, "downlinkEnabled": true }

--support
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10-10.0.22000-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.3a11
 Executable: C:\Users\User\appdata\local\programs\python\python310\scripts\meshtastic
 Python: 3.10.3 CPython MSC v.1929 64 bit (AMD64)
```

garthvh, 2022-04-08:
The client.Meshtastic.org site runs the latest version so is 1.3.5, channel presets changed in 1.3 and 1.2 devices are not compatible

closed 2022-04-08 by garthvh

rwcts, 2022-04-08:
That would do it. Thanks

rwcts, 2022-04-08:
I guess i should not have closed it so fast as the OLED is not displaying the proper speed either.

caveman99, 2022-04-08:
The web interface already uses the new protocol buffer definition for the 1.3 firmware. It's only a cosmetic issue.

Your 1.2 device is on LongSlow like you set it.

Protocol Buffers for 1.3: 

MidSlow = 3;

Protocol Buffers for 1.2:

Bw125Cr48Sf4096 = 3; (Which translates to LongSlow)

caveman99, 2022-04-08:
the OLED display bug is fixed in the 1.3 firmware. It's a display problem as well. the device is on LongSlow

rwcts, 2022-04-08:
I was going to mention i opened a bug on the 1.3 that was fixed but that sounds good. Thanks!