meshtastic/firmware #1371: [Bug]: inputbroker generate a reboot

amerinoj, 2022-04-12:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

1.3.5.595e5f4f

### Description

Some times when press any inputbroker_pin the ESP is reloader. 
`Preferences: { "screenOnSecs": 5, "phoneTimeoutSecs": 900, "lsSecs": 10, "region": "EU433", "chargeCurrent": "MA700", "gpsAccept2d": true, "extNotificationModuleEnabled": true, "extNotificationModuleOutputMs": 1500, "extNotificationModuleOutput": 2, "extNotificationModuleActive": true, "extNotificationModuleAlertMessage": true, "extNotificationModuleAlertBell": true, "positionFlags": 35, "inputbrokerPinB": 4, "inputbrokerPinPress": 14, "inputbrokerEventCw": "KEY_UP", "inputbrokerEventCcw": "KEY_DOWN", "inputbrokerEventPress": "KEY_SELECT", "updown1Enabled": true, "cannedMessageModuleEnabled": true, "cannedMessageModuleAllowInputSource": "upDown1", "cannedMessageModuleSendBell": true, "adcMultiplierOverride": 2.11 }
`

### Relevant log output

```Shell
a??:??:?? 61 Initialise the ESP32 bluetooth module
??:??:?? 61 Configuring the Device Information Service
??:??:?? 61 Configuring the Mesh bluetooth service
-------->>>abort() was called at PC 0x400858bb on core 1<<<<-----------------------------

ELF file SHA256: 0000000000000000

Backtrace: 0x4008fde7:0x3ffbea00 0x4009016d:0x3ffbea20 0x400858bb:0x3ffbea40 0x400859d5:0x3ffbea80 0x40139db6:0x3ffbeaa0 0x4013d025:0x3ffbedb0 0x40085c5a:0x3ffbede0 0x40147bb5:0x3ffbee20 0x40102fa0:0x3ffbee70 0x400d573e:0x3ffbee90 0x400d5a9e:0x3ffbeed0 0x400d4064:0x3ffbeef0 0x400f1a4b:0x3ffbef10 0x400f1a99:0x3ffbef30 0x400dc801:0x3ffbef50 0x401f07e2:0x3ffbef70 0x401f081d:0x3ffbef90 0x400dcd5d:0x3ffbefb0 0x400dcdfc:0x3ffbefe0 0x400813b1:0x3ffbf000 0x40082349:0x3ffbf020 0x4012d169:0x3ffbcc00 0x4012be25:0x3ffbcc20 0x40092439:0x3ffbcc40

Rebooting...
```

caveman99, 2022-04-27:
I defined new power states to keep the screen from turning off while operating the canned messages plugin. I most likely forgot a few transitions. I can reproduce the behaviour

garthvh, 2022-08-05:
Is this still an issue @caveman99 ? I am not getting a reboot loop when enabling canned messages on rak now, but I don't get a screen either.

garthvh, 2022-08-24:
Closing as I am not seeing this, please re-open if still having the issue on 1.3.39

closed 2022-08-24 by garthvh