meshtastic/firmware #1373: [Bug]: Canned message omit the last character when is sending

amerinoj, 2022-04-12:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

1.3.5.595e5f4f

### Description

When send a canned message the last character is omiter and the receiver don't display the last character.

caveman99, 2022-04-27:
When this happens, is canned_message_module_send_bell active or inactive? Looks like an Off-by-one Problem when assembling the payload packet in here. https://github.com/meshtastic/Meshtastic-device/blob/6c1dc0d71a8c578d483871ae5f07c3fec75e687f/src/modules/CannedMessageModule.cpp#L162-L182

amerinoj, 2022-04-27:
I haven't the device configured now .
But I  remember I had applied the configuration reported in the next issue .

https://github.com/meshtastic/Meshtastic-device/issues/1371

The send bell wasn't enabled in canned messages but yes was enable the extNotificationModuleAlertBell.

"extNotificationModuleAlertMessage": true, "extNotificationModuleAlertBell": true,

garthvh, 2022-08-15:
Confirmed that this is still an issue in 1.3.38

thebentern, 2022-10-17:
Still an issue?