meshtastic/firmware #1384: [enhancment]: add speed data to smart positioning

djscoutmaster, 2022-04-17:
### Category

Hardware Compatibility

### Hardware

Not Applicable

### Firmware Version

1.2

### Description

Once gps speed i formation is added ( #1379) then also add logic to not send any location update messages if speed is below 1mph.

this will aid in reducing congestion in the network when the use case are tracking traveling groups. (Bikes, cars, humans). 

When everyone spreads back out and mphs jump up, it can start reporting again.

Mph ticket
https://github.com/meshtastic/Meshtastic-device/issues/1379

mc-hamster, 2022-04-17:
How would knowing the speed in a short frequency help? Smart position requires moving a certain amount of distance in a certain amount of time, so it's a large frequency measure.

Removing the bug label. Existing feature works as expected.

mc-hamster, 2022-04-17:
Closing this. After conversation with @djscoutmaster, he suggested a bug rather than a need for an enhancement.

closed 2022-04-17 by mc-hamster