meshtastic/firmware #1390: please add firmware version into nodeinfo

rainerfritz, 2022-04-19:
Hi!

As we need here to know on which firmware each node is, I want to ask you to add firmware version information into the User section of the nodeinfo message. 

THX a lot
BR
Rainer

mc-hamster, 2022-04-19:
What's the use case for this?

rainerfritz, 2022-04-19:
We are running a HAM network with 40 to 60 nodes. You can have a look on it here: https://srv08.oevsv.at/mqtt/
We want to know on which firmware version each node is to react on the mqtt server for discrepances in the messages or if HAM ops update firmware we are able to keep track which node needs to be updated.

mc-hamster, 2022-04-19:
@rainerfritz Thanks!

Note to who ever implements this: It's ok to broadcast firmware version, but please don't add it to node info. It'll bloat that further which is already big. A separate plugin would be ideal.

caveman99, 2022-04-19:
The idea was to have it off by default, so the empty field would not take up space in the message.

mc-hamster, 2022-04-19:
so ... if nodedb is 80 entries and the version is 24 bytes (8bit.8bit.8bit) then that's 2k of ram at minimum.

rainerfritz, 2022-04-19:
What if we make a plugin which triggers a message to submit firmware data and all of other data needed. So no storing needed. Would cause a lot of airtime in the network, but we don't need that so often.

mc-hamster, 2022-04-19:
> What if we make a plugin which triggers a message to submit firmware data and all of other data needed. So no storing needed. Would cause a lot of airtime in the network, but we don't need that so often.

great idea ... please have that module be disabled by default

caveman99, 2022-04-19:
not that much airtime, cause that info is fairly static. A trigger every few hours would be enough. Battery Level is included already, and that's the only thing updating faster :-)

garthvh, 2022-08-05:
Might be useful to also request the protobuf version that is included in config now as well

garthvh, 2022-08-08:
Hey @rainerfritz will this work? 

https://github.com/meshtastic/Meshtastic-device/pull/1607

rainerfritz, 2022-08-08:
Hey Garth!
Yes that sounds great! Thx a lot for implementing it! Much appreciated!

thebentern, 2022-08-08:
This has been integrated
[image]

closed 2022-08-08 by garthvh