meshtastic/firmware #1392: [Bug]: Please add TBeam 0.7 in 433MHz

slavino, 2022-04-19:
### Category

Hardware Compatibility

### Hardware

T-Beam 0.7

### Firmware Version

latest

### Description

https://www.aliexpress.com/item/1005002051289452.html?spm=a2g0o.productlist.0.0.daa85f9d5XqGWz&algo_pvid=16ffe5ba-db4a-45eb-8882-5d1cc0fbfef1&

### Relevant log output

```Shell
Hello,
please add support for 433MHz of T-Beam 0.7 board.
```

caveman99, 2022-04-19:
T-Beam (0.7 and 1.1) already support 433 MHz. Just set it to region EU433. I'm rather surprised this 0.7 contraption is still on sale, no matter what frequency band and would recommend you skip it in favour of the 1.1 version with a decent charge controller.

closed 2022-04-19 by caveman99

mc-hamster, 2022-04-19:
> T-Beam (0.7 and 1.1) already support 433 MHz. Just set it to region EU433. I'm rather surprised this 0.7 contraption is still on sale, no matter what frequency band and would recommend you skip it in favour of the 1.1 version with a decent charge controller.

It's probably new old stock at this point. Agreed that 0.7 should be skipped.

slavino, 2022-04-20:
I have purchased pair of those TBeam-0.7 few years back.